iNavFlight/inav #5050: Multiple PID profiles on switch

ws-7, 2019-09-05:
## Current Behavior
Currently, using the CLI, it is possible to set multiple rate profiles in iNAV (https://github.com/iNavFlight/inav/blob/master/docs/Profiles.md). However, I haven't found a way to also set multiple PID profiles.

In Betaflight, I believe it is possible to set multiple PID profiles (https://www.reddit.com/r/Multicopter/comments/66tmdo/can_i_assign_2_different_pid_configurations_in/). 

## Desired Behavior
Add the option to iNAV (either through CLI or Configurator) to set multiple PID profiles, similarly to how it is currently already implemented in BetaFlight. 

## Suggested Solution
Would it be possible to fork the existing code regarding multiple PID profiles from Betaflight to iNAV? 

## Who does this impact? Who is this for?
This helps users who need to switch between multiple PID profiles during or between flights (e.g. aircraft with multiple means of propulsion like VTOL, people that want to fly with different types of batteries, people that want to change payloads and CG in between flights, etc.

MikeNH, 2019-09-10:
I'd like to add a request to be able to copy from one profile to another.  This would be a major troubleshooting tool if the copy ability could be done using the OSD menu or stick commands while in the field.  

If profile number one is not too bad but needs some trial and error troubleshooting,   those settings can be copied to profile number two and worked on without losing what works so far. 

Should this be a separate enhancement request?

michael-blue2, 2019-11-19:
Have we got any update on this one?

LaHob, 2019-12-02:
This sounds like a great feature, @DzikuVx  is there anything happening with this one? When you look at the global functions it appears to all be done already. Could you please tell me/us about the status?

DzikuVx, 2019-12-02:
Global functions are working, but PID on a switch will require extra work especially when you want to change them in flight

LaHob, 2019-12-02:
It's just that when you look from the project map to the global functions it appears to be all done. I was just wondering if there are any mods able to do this anytime soon. Do you know anything about that?

michael-blue2, 2019-12-02:
Hey Pawel, I am creating a tailsitter wing, fo rme to be able to hover it i need my P to be around 35-40 but as soon as i make the switch to forward flight the plane will get crazy with all that gain. So with a switch to change the profile will be awesome

LaHob, 2019-12-02:
> Hey Pawel, I am creating a tailsitter wing, fo rme to be able to hover it i need my P to be around 35-40 but as soon as i make the switch to forward flight the plane will get crazy with all that gain. So with a switch to change the profile will be awesome

This sounds like a great project, would love to be updated when it works.

JWC1304, 2019-12-04:
Same request, really hoping for a fix soon. Specific pid profiles would help my flying bird immensely. Flies great when flapping when transitioning into a glide mode (fixed wing) the pid profile is so off for this mode, it won't fly stable anymore.   

This sounds like a great project, would love to be updated when it works.

ws-7, 2019-12-05:
Great to see I'm not the only one with this request. I would like to second these other comments and again express my interest in the proposed feature!

DungAdumbs, 2020-07-17:
Working on VTOL transitions and could use this as well. 

Another nice feature would be to set "TPA" with a gain channel regardless of throttle position.   Might be easier to implement?