iNavFlight/inav #5051: Reassign motor output to servo?

keithluneau, 2019-09-05:
Hi, I'm setting up a CLRacing F4 Air V3 board into a fixed wing model. (https://cl-racing.myshopify.com/products/clracingf4-air-v3)

Currently there is two motor outputs, and a total of 4 servo outputs. The pinout for the board can be seen here: https://cdn.shopify.com/s/files/1/1737/4233/files/CLRACINGF4_AIR_V3.pdf?27

I will need a total of 5 servos and only 1 motor in this model, and the PWM pins on the receiver are not accessible so I'd like to run them all off the FC. Will it be possible to reassign one of the motor outputs to a servo instead? I've had a look at the target.c file for the board here: https://github.com/iNavFlight/inav/blob/master/src/main/target/CLRACINGF4AIR/target.c

Would it be as easy as changing line 31 from:
DEF_TIM(TIM8,  CH3,  PC8,   TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,    0, 0),
to:
DEF_TIM(TIM8,  CH3,  PC8,   TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    0, 0),

If so, is this still valid for compiling the firmware in Windows? https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md I've never done anything like this apart from playing around with Arduino a bit. Just making sure I'm not doing something wrong before I give it a shot... lol Thanks!

datamann, 2019-09-05:
@keithluneauTo my knowledge, you can't map a motor output to an servo output or vise versa because the timer run on different hertz, you risk servo burnout if you do.
I've been compiling INAV using docker but testet windows 10 linux subsystem a couple of weeks ago and found alot easyer.
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20with%20Linux%20Subsystem.md

keithluneau, 2019-09-05:
Even if I drive the ESC with 50Hz? I don't need anything special for this little plane in terms of throttle response... lol

BTW I spoke with Cheng Lin (of CLRacing) about it and he said this, "there are using the same Timer different channel. I think you can remap the Motor 2 to use as servo."

Does that mean I don't need to worry about the timer issue? Either way I'm fine with using 50Hz to stay on the safe side. 

Thanks for the tip on using Linux subsystem to compile, that sounds much easier! I did just set Cygwin and the other related stuff and compiled and it seems to have worked. I'll give that method a try though for the future, sounds a lot simpler to me...

keithluneau, 2019-09-06:
Just a quick update...

This evening I tried this out. I changed line 31 as above from "MOTOR" to "SERVO". This changed what was normally "Motor 1" in iNav to a servo output. All servos including the "new" one worked as expected. (I did set ESC to 50Hz to be safe) However, the ESC plugged into what was originally Motor 2, now Motor 1, didn't work. 

I flashed the original firmware back on the board and used the default airplane mixer, and set the ESC to 50Hz to make sure it would work like that. It did, no issues. So I then changed line 32 from "MOTOR" to "ESC", which changed what was Motor 2 in iNav to a servo output. I left the ESC in Motor 1 as normal. Again I set ESC to 50Hz. This time, the ESC worked normally, but none of the servos worked in ANY servo ports.  I unplugged the servo from what was originally Motor 2 and set the ESC back to 400Hz, it still worked, the servos were all still not working.

So at this point I can only assume there's something deeper going on in the code that's breaking when I change these pin assignments. I'm already outside my comfort zone and my abilities, so I'm calling it quits unless someone else can shed some light on what I would need to get it working. For now, I'll sacrifice the tilt servo and fly with just pan. ;)

Fraggs150, 2020-08-13:
I believe this feature request for the motor output to be reassigned to a servo output has been implimented on the Noxe v1 upgraded version of this flight controller (FLYWOO F411) https://github.com/iNavFlight/inav/issues/5867
Hopefully it can also be implimented on this version of the Noxe too, it would open up the possibility of lots more models being used having the ability  to use rudder,aileron,elevator and motor.
Cheers
 Stu

closed 2022-05-07 by DzikuVx

zvikaf, 2022-07-25:
Was there any change, or any information on how to assign servo to the second moter PWM output ? 
tried recompiling with assigning different timer with strange outcome..