iNavFlight/inav #5052: HAKRC F4 and a compass

grafvadim, 2019-09-06:
Good day.
There is an FC HAKRC F4 and a BN-880 module (GPS and compass) and CrossFire Nano RX.
Connected GPS to RX6, TX6. CrossFire on RX3, TX3, Works. But where is the compass?
I tried on TX1, RX1. INAV does not see the compass. I tried on Crossfire nano RX on the ports BST SDA and BST SDL does not see the compass.
I tried to solder on a compass on RX3, TX3, and crossFire on RX1, TX1, I see neither a compass, nor crossfire.
Already I don’t know what to do, can anyone know how to solve the problem?

DzikuVx, 2019-09-06:
Compass has to be connected via I2C bus with pads SCL to SCL and SDA to SDA. Refer to the flight controller documentation where I2C pins are broken out.
On the BN880 IIRC SCL is marked as C and SDA as D

closed 2019-09-06 by DzikuVx

grafvadim, 2019-09-06:
https://ae01.alicdn.com/kf/HTB12EzUOmrqK1RjSZK9q6xyypXax.jpg
The fact of the matter is that there are no SDL and SDA contacts on the FC board.