iNavFlight/inav #5053: Inav gps mission oscilation in altitude

purpletreee, 2019-09-06:
Hello Guys, 
I have a problem. In angle mode my nano talon is total stable and no oscilations. But when I start a gps mission it works, it follows the path, but there are always oscilations in the height. I set the height to 120 meters and it oscilates from 116 to 125 meters. And the motor throttle is oscilating to probably to correct the height. All this oscilations are bad for my rx signal, because when ever the plane is pitching up the angle of the antenna doesnt match my antenna I think. I am not sure if this has something to do with PIDs or PIFFs, and so I dont want to mess up any settings.

wx4cb, 2019-09-06:
try adjusting your cruise throttle a little bit, also. make sure you run it through an auto tune etc

purpletreee, 2019-09-06:
thank you i will try that tomorrow

closed 2019-11-19 by stale[bot]