iNavFlight/inav #5054: Compass not calibrating

nosyninja, 2019-09-08:
## Current Behavior
INav Client fails to calibrate compass during the 30 second window. X Y Z all remain set to 0.
HMC5883 on SPRacing F3 Delux Flight Controller.

## Steps to Reproduce

1. Calibrate compass.
2. Save

## Expected behavior
X Y Z values set to numbers other than 0

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:

# version
# INAV/SPRACINGF3 2.2.1 Jul  3 2019 / 21:50:13 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

sammichael25, 2019-09-09:
I have the exact same issue but i'm using a Matek f405 AIO and M8M with HMC5883

nosyninja, 2019-09-11:
My workaround -

The quad can not be armed, because the compass won't calibrate. To get round this, I installed cleanflight configurator, flashed the flight controller, and then took the magnetometer x, y, z values from cleanflight. I then flashed inav, and set the magnetometer values (by cli), using the cleanflight values. This is not as accurate the 6 point calibration in inav, but at least it gets me flying.

jaysinism, 2020-01-16:
I am also having the same issue with compass not calibrating. It counts down the 30 seconds the first time but all the Mag numbers stay zero. I am using iNav 2.3.3 with the latest firmware for Kakute F4 v2 and I am using the Beitian BN-880 GPS module. Is there a way we can manually set it? It was working before with the older iNav

flythebluesky, 2020-04-29:
I'm having the same issues with Kakute F7 w/Matek M8Q-5883. Also seeing 0, 0, 0 on the calibration but every now and then the calibration numbers appear and the disappear on next attempt at calibration. The magnetometer has been swapped out. Same problem. Also tried different i2c_speed settings. After installing 5.1 k ohm pullup resistors it seemed to work in Betaflight (with pullup setting ON), but I only tried that once. Is there any way to verify the i2c is working correctly?

nosyninja, 2020-04-29:
I've given up with the mag/compass now, switched it off. I'll try a GPS with built in compass next.

ShikOfTheRa, 2020-05-04:
Same on a new build
Kakute F7 mini w/Matek M8Q-5883. Also seeing 0, 0, 0 on the calibration but every now and then the calibration numbers appear and the disappear on next attempt"
Never had this issue in year of using iNAVverxsions  up to v 2.2 (which will try next)

digitalentity, 2020-05-04:
Does the compass return valid readings in the Sensors tab? Do readings change when you move the compass or put a piece of metal next to it?

flythebluesky, 2020-05-04:
I have unloaded iNav, so I don't know. I honestly don't think iNav works with Kakute F7. I have tried iNav with 2 different Kakute F7 boards. I have tried M8Q-5883 and BN-880 - and I bought 2 of each just in case I received bad hardware. With both GPS/mag iNav autodetects the magnetometers and it knows when the magnetometer is on or off. I have tried many different board settings and orientations. Nothing has worked. The compass readings are never correct.

flythebluesky, 2020-05-04:
Also, I have tried from a fresh setup - configuring magnetometer only. It hasn't worked at all.

digitalentity, 2020-05-04:
KAKUTE F7 works with the magnetometer, this is verified by several users. 
If INAV detects the magnetometer, but compass traces on Sensors tab don't move, this is a hardware issue (most likely due to I2C pullups or power).

ShikOfTheRa, 2020-05-04:
Yes - seems normal.
Yes to metal nearby.
Have set orientation so it is correct - i.e. don't get that strange wobbly image rotating. 
Its F7 Mini although shouldn't make difference.
Once get the calibration value in, it seem to be OK. I a have a load of testing and config to try first.
Always Get 45 I2C error at connect, but its not counting up at all after that.  Tried I2c 100-800HZ with no difference. Would expect a few issue if I2C pullups issues.  I don't know what is normal, but I  don't worry too much unless its rising.
It's quiet strange. I will try a few other things to see if can pin it down further. 
 
> Does the compass return valid readings in the Sensors tab? Do readings change when you move the compass or put a piece of metal next to it?

digitalentity, 2020-05-04:
This is odd. When you get no values in the Configurator, can you go to CLI and check if `magzero` is set there? This might be a Configurator issue or firmware issue and I'd like to differentiate between the two

ShikOfTheRa, 2020-05-04:
Good idea

I noted the values I had set in the GUI
Ran calibration - all values zero in GUI after calibration
Values for magzero are present in the CLI - different to before and approximately the values that were there before.

Think that helps pinpoint area. 
2020-05-04 @ 17:45:57 -- Running - OS: Windows, Chrome: 75.0.3770.142, Configurator: 2.4.1
2020-05-04 @ 17:45:57 -- Loaded release information from GitHub.
2020-05-04 @ 17:46:54 -- Serial port successfully opened with ID: 1
2020-05-04 @ 17:46:54 -- MultiWii API version received - 2.4.0
2020-05-04 @ 17:46:54 -- Flight controller info, identifier: INAV, version: 2.4.0
2020-05-04 @ 17:46:54 -- Running firmware released on: Mar 9 2020 21:01:51
2020-05-04 @ 17:46:54 -- Board: KF7M, version: 0