iNavFlight/inav #5055: Wizard X220s

LawrenceLDJr, 2019-09-09:
## Current Behavior

Brand new X220s, BN 880, R9 Slim+ and Wolfwhoop VTX. Flips over and crashes to ground  
## Steps to Reproduce

1.Flashed Omnibus F4 V3 LED M5 SS M6 Soldered LED wire to M5 soldered smart audio M6
2.Preset Preformance
3.All sensors BLUE all pre flight checked green
4.Take off and fly , using alt hold it randomly flips over and hits ground. Alt Hold enabled at all times 

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: Omnibus F4 V3 (Eachine /Airbot who knows)
- INAV version string:

DzikuVx, 2019-09-09:
@LawrenceLDJr please provide blackbox log and DVR of the event

FlyR9, 2019-09-13:
> Just reflashed, was gonna try DShot 300 since there are no warnings. Does not preformance use Multishot? Gonna give Multishot another whirl then. Crash that destroyed 2 motors, frame, and a omnibus F4 V6. Screen went black but was only 20m away but 50m up. It just shut down and dropped onto cement. Was my first quad with INav, preformance preset wich uses multishot. I don't blame INav for this but the quad never had a problem in betaflight. Gonna enable airmode this time. I didn't use airmode since I like to take off and land and some cliffhanger stuff was happening. I can scratch the landings for the alt hold, pos hold and waypoint missions. Was great til that max throttle min throttle episode. Brand new X220s $140, smart audio capable vtx $20, R9MM $20, Runcam 5 $100, BN880 $20. I loose this I'm getting a Mavic.

Really sad to hear that, but if i have good understand you set the performance preset wich use the multishot protocol? If it's that you need to calibrate the max/min throttle manually and this can explain these issues i think. I cant blam you for that but the first tests of any build should be the lightest possible (without action cam and with a light battery it's better) and it should be on the grass.. all of this to reduce the damage on you're build int he case of a crash. However i think we can highlight something wrong in these inav experiences, im actually looking for make an inav build and i cant find detaillied videos of inav something to explain it for the beginners like Joshua Bardwell or a lot of others on betaflight.. The only ones i've find is on Russian, if someone know a full setup tutorial of inav in English it could help me a lot thank you.

LawrenceLDJr, 2019-09-13:
ESC from a Tyro99 was failing and did fail using INav. So judgement not in yet. 50m drop was the Tyro99 but had omnibus F4 V6 flight controller. Had insurance on the omnibus F4 V6 and was replaced. New quad with omygawd frame Tyro99 ESC omnibus F4 V6 Tyro99 motors. Flip overs big time and finally the ESC fried. Bought the X220 $140, to cheap to pass up. Flashed INav, began mowing and excavating my backyard oneshot multishot finally watched the Pawel Palowsiky and started using DShot 600 and for awhile almost perfect alt hold, pos hold and finally waypoint missions working beautifully. Then crazy max throttle min throttle event. Calibrating the non digital protocols is something I've never done so this time I will try multishot properly calibrated. Pawel states Dshot 300 works without issue and doesn't have a warning when you choose it. Still I'm gonna take teckel12s advice multishot only this time it will be calibrated. This attempt will be the X220s. Read Oscar L esc tutorial and learned about calibration of analog protocols.

LawrenceLDJr, 2019-09-13:
Was beating myself up about the ESC calibration issue, if ya read the INav setup pages thoroughly you would have calibrated your ESCs to use analog protocols (silly INJUN). Just read them and re read them and and I don't think it is mentioned any where. New motors and a Tarsier in the mail today. Also have a Typhoon ESC on the way to match up with the Omnibus F4 V6. I will be using Wizard X220S this time around. I can't wait to run the way point mission around my local park. THX for your input. Still gonna post results.

digitalentity, 2019-09-14:
Hard to tell anything without a blackbox log.

DSHOT (especially high speed) is very picky about wiring quality. DSHOT150 generally is as reliable as Multishot (in terms of signal quality). Full throttle in AltHold is generally associated with vibration causing high-frequency acceleration beyond 8G. Vibration issue can only be solved mechanically - by vibration damping your FC and/or motors.

teckel12, 2019-09-15:
@LawrenceLDJr I suggested multishot as you will NEVER have a problem due to multishot, NEVER. DSHOT, on the other hand, absolutely CAN cause a problem, even DSHOT 300. And why would you use DSHOT 300 when Multishot is faster and more reliable?

I suggested to enable airmode during the entire flight because it may be linked to alt hold problems. Someone reported that with airmode enabled alt hold worked correctly, but with airmode disabled, alt hold went crazy.

So... I'm suggesting Multishot and airmode, to see of that resolves your problem. Any problem you had previously wasn't the cause of multishot, it's FAR more stable than DSHOT.

LawrenceLDJr, 2019-09-16:
Flights today were excellent... multishot this time but calibrated ESCs. Take offs and landings in stab mode only. Alt hold with airmode worked excellent, pos hold and air mode was perfect with no toilet bowl. The destroyed quad I made it clear I didn't believe INav or multishot was at fault. That was the status of my settings when it happened. Reason for Dshot was something I read about errors making it into the PID loop with analog protocols. Digital won't use the corrupted data. Probably so miner ya don't see it I'm sure but that was the reason. 8 packs today without an issue. Using opentx to lock throttle at 1500 in alt hold and I think alt hold is perfect. It drifts downward at high speed and hard menuevers, but goes right back to alt you set when throttle locked at 1500. I'm having lotta fun...

FlyR9, 2019-09-16:
> Flights today were excellent... multishot this time but calibrated ESCs. Take offs and landings in stab mode only. Alt hold with airmode worked excellent, pos hold and air mode was perfect with no [...]

Thanks for sharing your results i hope it help other peoples ^^

LawrenceLDJr, 2019-09-17:
Conclusion. INav didn't like my fleet of beat up betaflight quads. The new X220S wouldn't stop flipping out because I think the un calibrated ESCs when using multishot and oneshot 125. Flew very well on DShot 600, but had one incident that brought me here. Flies excellent with multishot now that I've calibrated ESCs. Over 30 batterys without incident. Big thx for support Pawel, FlyR9 and teckel12.

teckel12, 2019-09-19:
@LawrenceLDJr I would highly suggest using Multishot.  It's as fast as DSHOT 600 and as reliable as DSHOT 150.  And if you're a fast-twitch elite pilot where only DSHOT 1200 will do, what are you using INAV for?  Betaflight and INAV serve different clientele.

LawrenceLDJr, 2019-09-21:
I would consider myself INav pilot although I've only flown it a month. I'm well aware of the speeds of all the protocols. I would suggest warning new INav pilots about flying without calibration of the analog protocols. Most reliability is what I seek not speed. I'm permanent stab alt hold settings with this quad. With the help of this and other forums I'm achieved what I was looking for. Multishot after calibration has 100 problem free flights now. Cheers...!

teckel12, 2019-09-21:
@LawrenceLDJr Calibrating the ESCs is ways a must, I'd do that even if I was using DSHOT. There's about 500 warnings that could be given. It's the user's responsibility to know what to do. If I would give any warning it would be to not use DSHOT.

closed 2019-09-21 by LawrenceLDJr