iNavFlight/inav #5056: LED WIRING ORDER MODE NOT WORKING

pmkone, 2019-09-09:
## Current Behavior
When trying to setup LEDs, enabling wiring order mode does not allow me to number the "boxes". They just stay blank (highlighted purple though)

## Steps to Reproduce

1. Enable Led Strip
2. Go To Led Strip GUI setup
3. Select Wiring Order Mode (or any combination of options) and try numbering grid

## Expected behavior
When Selecting Wiring order mode and clicking or highlighting multiple "boxes" on the "grid", the box should get a number (representing the led position in the shift register)

## Additional context
CLI LED command missing

# version
# INAV/OMNIBUSF7 2.2.1 Jul  3 2019 / 22:21:59 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# resource
IO:
----------------------
A00: FREE 
A01: BARO CS
A02: FREE 
A03: FREE 
A04: SPI1 CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: FREE 
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: MPU CS
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B02: FREE 
B03: FREE 
B04: FREE 
B05: FREE 
B06: FREE 
B07: FREE 
B08: FREE 
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: FREE 
C04: FREE 
C05: FREE 
C06: SERIAL6 UART TX
C07: SERIAL6 UART RX
C08: FREE 
C09: FREE 
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
[... 1083 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 1300
set battery_capacity_warning = 650
set battery_capacity_critical = 390
set battery_capacity_unit = MAH

skorokithakis, 2019-10-02:
LED is not enabled yet on a Kakute F7 (see #4150).

DzikuVx, 2020-04-14:
LED is not enabled on OmnibusF7

closed 2020-04-14 by DzikuVx