iNavFlight/inav #5058: After Loading Arduplane and re-flashing back to INAV , No SBUS Seen From Receiver.

Xigxagz, 2019-09-10:
[INAV Dump.txt](https://github.com/iNavFlight/inav/files/3593082/INAV.Dump.txt)

Hello I'm.looking for an opinion/help on my issue.
I have 3 F405 Wing boards and for the life of me, I can't get 2 of them to show any movements in the "Receiver" portion of INAV setup.

I'm using the latest firmware on everything (configurator and firmware).
I'm using the same DragonLink Nano receiver to do the testing as I swap the F405's out my testing set-up.
I'm getting powering to my DL Nano receiver through the two problem boards also.

One F405 works fine, I'm able to see the stick movements in INAV and fire up my test motor, but the other two act like there's no SBUS receiver connected and the board stays in "Failsafe" because of the fact that there's no SBUS indicated in the INAV set-up.

I should mention that I loaded ArduPlane on the two boards that indicate no SBUS after I've flashed back to INAV. When flashing back, I've done a "full chip erase" every time also. The F405 that works has never had ArduPlane flashed onto it.
I've tried re-flashing INAV, and also have done a "diff all" copy and paste in CLI from my good board onto the two faulty ones, with no success.

All "Pre-Arming" checks are GREEN, except the Failsafe parachute is RED

So, I'm kinda wondering that there's a firmware snafu that won't shake out because I flashed ArduPlane on the two problem boards.
Thank you for any help !
John

datamann, 2019-09-10:
Did you try Betaflight to see if SBUS works there?

Xigxagz, 2019-09-10:
No, I have not. I will give that a try and see. Thanks !

Xigxagz, 2019-09-11:
Thanks Stig...  After re-flashing the latest INAV file via the Betaflight configurator ( from a saved INAV file on my desktop) that fixed my issue.Is this a known problem from others who have flashed ArduPlane and then have decided to come back to INAV ?? The fact that I had to flash INAV using Betaflight to cure the problem seems a bit strange to me; but whatever it takes. I just wish that I would've known this sooner.Thanks again for the suggestion...you saved me.  :)John

DzikuVx, 2019-09-13:
The solution was probably to flash with full chip erase

closed 2019-09-13 by DzikuVx

Xigxagz, 2019-09-13:
Hello Pawel ..  I just wanted to clarify, that I DID use full chip erase many times when re-flashing with the INAV configurator, but still the INAV configurator would not recognize the SBUS coming from my receiver. I also did a fresh install of the INAV configurator to make sure that it was not corrupted in some way.
The only thing that finally solved it was doing the full chip erase and flashing my local INAV firmware file using the BetaFlight configurator ... and then after that I could connect to the INAV configurator and find SBUS being recognized.
Thank you !John