iNavFlight/inav #5059: Using MAVLink to view telemetry in QGroundControl

ws-7, 2019-09-10:
The INAV missions page (https://github.com/iNavFlight/inav/wiki/iNavFlight-Missions) shows that INAV supports MAVLink interfacing with QGroundControl. Anyone here who can help me with this? 

Eventually my plan is to use a radio telemetry link (probably Dragonlink) for my telemetry signal, and visualize my flight path on a map (which QGroundControl should be able to do). At first, though, I want to try and get the INAV telemetry through MAVLink working over USB. But does it even work over USB? I've tried setting the USB Port settings in the INAV configurator to 115200 baud MAVLINK but so far can not get a connection to QGroundControl.

Anyone who has experience with using the combination of INAV, MAVLink and QGroundControl who can point me in the right direction? Thanks!

wx4cb, 2019-09-10:
use MWPtools on a laptop. works great. http://www.github.com/stronnag/mwptools

stronnag, 2019-09-11:
The wiki has been updated to reflect the claim that contemporary qgroundcontrol no longer processes iNav mavlinbk telemetry (not wihstanding the fact that qgroundcontrol's telemetry log records valid Mavlink data (which is processed as valid mavlink telemetry by [mwp](https://github.com/stronnag/mwptools)

whoim2, 2019-10-04:
What version of qgroudcontrol work?

stronnag, 2019-10-05:
No idea, something from c. 3.5 years ago did, once.

closed 2019-12-18 by stale[bot]