iNavFlight/inav #5063: Ublox M8Q-5883 mag not working on matek 405std

robohippy, 2019-09-11:
anyone help with mag issue on matek 405std and a matek gps, the mag isnt working
ive changed the i2c bus speed to 200khz and it still does not work
im using a Ublox M8Q-5883
mag set to QMC 5883
the gps works but not the mag

robohippy, 2019-09-13:
solved sda and sca wired wrong way round

closed 2019-09-13 by robohippy

Z-Vladus, 2022-11-09:
Hello guys.
Have my Matek 405std + Matek M8Q-5883 connection problem. 
I think I solved it. 
Try soldering +5V of your GPS module to "4V5" pad of your flight controller instead of 5V pad.
I suppose baro is powered by 4v5 bus, as the baro works right away if the USB is connected. I think all other I2C devices should be powered the same way.