iNavFlight/inav #5069: INAV 2.2.1 Failsafe & Lockout Using Matek F722

JDFPV, 2019-09-14:
This happened to my friend on the 5th flight with Ar 900 Wing. I launched the plane for him, the wing tracked out as normal. He switched to RTH mode while he was getting cozy in his seat and putting goggles on. Once he was ready to fly he switched to HOR mode and FS! popped up on the screen. He tried everything you can imagine to regain control and nothing worked. Complete Lockout. The only thing the flight controller responded to was the camera switcher. He could change cameras. The rssi was good. The pilot did say he was still in the tuning process is why the plane was flying so crazy in rth. But how and why did the failsafe engage and the lockout?? 

Components used: 
Sonicmodell AR Wing 900mm Wingspan PNP Frsky R9MM 
AKK FX2-Dominator 
Beitian Dual BN-220 GPS 
Matek Systems F722-WING
 RunCam Split 3 Micro 1080P 60fps HS1193 Foxeer Night Wolf 
(2) TitanPower 4S Lion 3500mAh 

DVR Footage from this flight and crash
https://youtu.be/AX3EzFXBvL4

No blackbox data 

# dump # version # INAV/MATEKF722SE 2.2.1 Jul 3 2019 / 22:44:40 (a6d847482) # GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074] # resources # mixer mmix reset mmix 0 1.000 0.000 0.000 0.000 mmix 1 1.000 0.000 0.000 0.000 # servo mix smix reset smix 0 3 0 50 0 -1 smix 1 3 1 50 0 -1 smix 2 4 0 -50 0 -1 smix 3 4 1 50 0 -1 # servo servo 0 1000 2000 1500 100 servo 1 1000 2000 1500 100 servo 2 1000 2000 1500 100 servo 3 1000 2000 1500 -100 servo 4 1000 2000 1500 100 servo 5 1000 2000 1500 100 servo 6 1000 2000 1500 100 servo 7 1000 2000 1500 100 # logic logic 0 0 0 0 0 0 0 0 logic 1 0 0 0 0 0 0 0 logic 2 0 0 0 0 0 0 0 logic 3 0 0 0 0 0 0 0 logic 4 0 0 0 0 0 0 0 logic 5 0 0 0 0 0 0 0 logic 6 0 0 0 0 0 0 0 logic 7 0 0 0 0 0 0 0 logic 8 0 0 0 0 0 0 0 logic 9 0 0 0 0 0 0 0 logic 10 0 0 0 0 0 0 0 logic 11 0 0 0 0 0 0 0 logic 12 0 0 0 0 0 0 0 logic 13 0 0 0 0 0 0 0 logic 14 0 0 0 0 0 0 0 logic 15 0 0 0 0 0 0 0 # feature feature -THR_VBAT_COMP feature -VBAT feature -TX_PROF_SEL feature -BAT_PROF_AUTOSWITCH feature -MOTOR_STOP feature -SOFTSERIAL feature -GPS feature -TELEMETRY feature -CURRENT_METER feature -3D feature -RX_PARALLEL_PWM feature -RX_MSP feature -RSSI_ADC feature -LED_STRIP feature -DASHBOARD feature -BLACKBOX feature -TRANSPONDER feature -AIRMODE feature -SUPEREXPO feature -VTX feature -RX_SPI feature -PWM_SERVO_DRIVER feature -PWM_OUTPUT_ENABLE feature -OSD feature -FW_LAUNCH feature VBAT feature MOTOR_STOP feature SOFTSERIAL feature GPS feature TELEMETRY feature CURRENT_METER feature LED_STRIP feature BLACKBOX feature AIRMODE feature PWM_OUTPUT_ENABLE feature OSD # beeper beeper RUNTIME_CALIBRATION beeper HW_FAILURE beeper RX_LOST beeper RX_LOST_LANDING beeper DISARMING beeper ARMING beeper ARMING_GPS_FIX beeper BAT_CRIT_LOW beeper BAT_LOW beeper GPS_STATUS beeper RX_SET beeper ACTION_SUCCESS beeper ACTION_FAIL beeper READY_BEEP beeper MULTI_BEEPS beeper DISARM_REPEAT beeper ARMED beeper SYSTEM_INIT beeper ON_USB beeper LAUNCH_MODE beeper CAM_CONNECTION_OPEN beeper CAM_CONNECTION_CLOSED # map map AETR # serial serial 20 1 115200 115200 0 115200 serial 0 1 115200 115200 0 115200 serial 1 64 115200 115200 0 115200 serial 2 2 115200 115200 0 115200 serial 3 2048 115200 115200 0 115200 serial 5 1024 115200 115200 0 115200 serial 30 0 115200 115200 0 115200 # led led 0 0,0::C:0 led 1 0,0::C:0 led 2 0,0::C:0 led 3 0,0::C:0 led 4 0,0::C:0 led 5 0,0::C:0 led 6 0,0::C:0 led 7 0,0::C:0 led 8 0,0::C:0 led 9 0,0::C:0 led 10 0,0::C:0 led 11 0,0::C:0 led 12 0,0::C:0 led 13 0,0::C:0 led 14 0,0::C:0 led 15 0,0::C:0 led 16 0,0::C:0 led 17 0,0::C:0 led 18 0,0::C:0 led 19 0,0::C:0 led 20 0,0::C:0 led 21 0,0::C:0 led 22 0,0::C:0 led 23 0,0::C:0 led 24 0,0::C:0 led 25 0,0::C:0 led 26 0,0::C:0 led 27 0,0::C:0 led 28 0,0::C:0 led 29 0,0::C:0 led 30 0,0::C:0 led 31 0,0::C:0 # color color 0 0,0,0 color 1 0,255,255 color 2 0,0,255 color 3 30,0,255 color 4 60,0,255 color 5 90,0,255 color 6 120,0,255 color 7 150,0,255 color 8 180,0,255 color 9 210,0,255 color 10 240,0,255 color 11 270,0,255 color 12 300,0,255 color 13 330,0,255 color 14 0,0,0 color 15 0,0,0 # mode_color mode_color 0 0 1 mode_color 0 1 11 mode_color 0 2 2 mode_color 0 3 13 mode_color 0 4 10 mode_color 0 5 3 mode_color 1 0 5 mode_color 1 1 11 mode_color 1 2 3 mode_color 1 3 13 mode_color 1 4 10 mode_color 1 5 3 mode_color 2 0 10 mode_color 2 1 11 mode_color 2 2 4 mode_color 2 3 13 mode_color 2 4 10 mode_color 2 5 3 mode_color 3 0 8 mode_color 3 1 11 mode_color 3 2 4 mode_color 3 3 13 mode_color 3 4 10 mode_color 3 5 3 mode_color 4 0 7 mode_color 4 1 11 mode_color 4 2 3 mode_color 4 3 13 mode_color 4 4 10 mode_color 4 5 3 mode_color 5 0 9 mode_color 5 1 11 mode_color 5 2 2 mode_color 5 3 13 mode_color 5 4 10 mode_color 5 5 3 mode_color 6 0 6 mode_color 6 1 10 mode_color 6 2 1 mode_color 6 3 0 mode_color 6 4 0 mode_color 6 5 2 mode_color 6 6 3 mode_color 6 7 6 mode_color 6 8 0 mode_color 6 9 0 mode_color 6 10 0 # aux aux 0 0 2 1700 2100 aux 1 2 0 900 1300 aux 2 3 0 1300 1650 aux 3 3 4 1300 2100 aux 4 8 6 900 1600 aux 5 35 4 1300 2100 aux 6 10 5 1300 2100 aux 7 28 7 1300 2100 aux 8 18 1 1700 2100 aux 9 37 8 900 1300 aux 10 38 3 1700 2100 aux 11 0 0 900 900 aux 12 0 0 900 900 aux 13 0 0 900 900 aux 14 0 0 900 900 aux 15 0 0 900 900 aux 16 0 0 900 900 aux 17 0 0 900 900 aux 18 0 0 900 900 aux 19 0 0 900 900 # adjrange adjrange 0 0 0 900 900 0 0 adjrange 1 0 0 900 900 0 0 adjrange 2 0 0 900 900 0 0 adjrange 3 0 0 900 900 0 0 adjrange 4 0 0 900 900 0 0 adjrange 5 0 0 900 900 0 0 adjrange 6 0 0 900 900 0 0 adjrange 7 0 0 900 900 0 0 adjrange 8 0 0 900 900 0 0 adjrange 9 0 0 900 900 0 0 adjrange 10 0 0 900 900 0 0 adjrange 11 0 0 900 900 0 0 # rxrange rxrange 0 1000 2000 rxrange 1 1000 2000 rxrange 2 1000 2000 rxrange 3 1000 2000 # temp_sensor temp_sensor 0 0 0 0 0 0 temp_sensor 1 0 0 0 0 0 temp_sensor 2 0 0 0 0 0 temp_sensor 3 0 0 0 0 0 temp_sensor 4 0 0 0 0 0 temp_sensor 5 0 0 0 0 0 temp_sensor 6 0 0 0 0 0 temp_sensor 7 0 0 0 0 0 # wp #wp 0 invalid wp 0 0 0 0 0 0 0 wp 1 0 0 0 0 0 0 wp 2 0 0 0 0 0 0 wp 3 0 0 0 0 0 0 wp 4 0 0 0 0 0 0 wp 5 0 0 0 0 0 0 wp 6 0 0 0 0 0 0 wp 7 0 0 0 0 0 0 wp 8 0 0 0 0 0 0 wp 9 0 0 0 0 0 0 wp 10 0 0 0 0 0 0 wp 11 0 0 0 0 0 0 wp 12 0 0 0 0 0 0 wp 13 0 0 0 0 0 0 wp 14 0 0 0 0 0 0 wp 15 0 0 0 0 0 0 wp 16 0 0 0 0 0 0 wp 17 0 0 0 0 0 0 wp 18 0 0 0 0 0 0 wp 19 0 0 0 0 0 0 wp 20 0 0 0 0 0 0 wp 21 0 0 0 0 0 0 wp 22 0 0 0 0 0 0 wp 23 0 0 0 0 0 0 wp 24 0 0 0 0 0 0 wp 25 0 0 0 0 0 0 wp 26 0 0 0 0 0 0 wp 27 0 0 0 0 0 0 wp 28 0 0 0 0 0 0 wp 29 0 0 0 0 0 0 wp 30 0 0 0 0 0 0 wp 31 0 0 0 0 0 0 wp 32 0 0 0 0 0 0 wp 33 0 0 0 0 0 0 wp 34 0 0 0 0 0 0 wp 35 0 0 0 0 0 0 wp 36 0 0 0 0 0 0 wp 37 0 0 0 0 0 0 wp 38 0 0 0 0 0 0 wp 39 0 0 0 0 0 0 wp 40 0 0 0 0 0 0 wp 41 0 0 0 0 0 0 wp 42 0 0 0 0 0 0 wp 43 0 0 0 0 0 0 wp 44 0 0 0 0 0 0 wp 45 0 0 0 0 0 0 wp 46 0 0 0 0 0 0 wp 47 0 0 0 0 0 0 wp 48 0 0 0 0 0 0 wp 49 0 0 0 0 0 0 wp 50 0 0 0 0 0 0 wp 51 0 0 0 0 0 0 wp 52 0 0 0 0 0 0 wp 53 0 0 0 0 0 0 wp 54 0 0 0 0 0 0 wp 55 0 0 0 0 0 0 wp 56 0 0 0 0 0 0 wp 57 0 0 0 0 0 0 wp 58 0 0 0 0 0 0 wp 59 0 0 0 0 0 0 # osd_layout osd_layout 0 0 2 3 V osd_layout 0 1 24 1 V osd_layout 0 2 0 0 H osd_layout 0 3 8 6 V osd_layout 0 4 8 6 H osd_layout 0 5 23 8 H osd_layout 0 6 23 9 H osd_layout 0 7 1 1 V osd_layout 0 8 10 12 V osd_layout 0 9 1 2 H osd_layout 0 10 8 6 H osd_layout 0 11 24 2 V osd_layout 0 12 24 12 V osd_layout 0 13 0 6 V osd_layout 0 14 1 2 V osd_layout 0 15 24 6 V osd_layout 0 16 2 10 H osd_layout 0 17 2 11 H osd_layout 0 18 2 12 H osd_layout 0 19 15 1 H osd_layout 0 20 1 0 V osd_layout 0 21 18 0 V osd_layout 0 22 14 2 V osd_layout 0 23 11 3 V osd_layout 0 24 12 0 V osd_layout 0 25 23 5 V osd_layout 0 26 24 7 H osd_layout 0 27 3 5 H osd_layout 0 28 11 11 V osd_layout 0 29 1 12 V osd_layout 0 30 0 10 V osd_layout 0 31 1 3 H osd_layout 0 32 12 1 H osd_layout 0 33 24 3 V osd_layout 0 34 10 1 V osd_layout 0 35 24 11 V osd_layout 0 36 1 5 H osd_layout 0 37 1 6 H osd_layout 0 38 1 7 H osd_layout 0 39 1 5 H osd_layout 0 40 1 11 V osd_layout 0 41 1 8 H osd_layout 0 42 1 7 H osd_layout 0 43 0 0 H osd_layout 0 44 0 0 H osd_layout 0 45 0 0 H osd_layout 0 46 1 10 H osd_layout 0 47 3 7 H osd_layout 0 48 23 7 H osd_layout 0 49 23 6 H osd_layout 0 50 0 0 H osd_layout 0 51 12 2 H osd_layout 0 52 12 2 H osd_layout 0 53 12 1 H osd_layout 0 54 12 1 H osd_layout 0 55 1 8 H osd_layout 0 56 2 12 H osd_layout 0 57 2 12 H osd_layout 0 58 2 12 H osd_layout 0 59 2 12 H osd_layout 0 60 2 12 H osd_layout 0 61 2 12 H osd_layout 0 62 2 10 H osd_layout 0 63 2 11 H osd_layou [... cut ...] e = 20 set yaw_rate = 20 set manual_rc_expo = 70 set manual_rc_yaw_expo = 20 set manual_roll_rate = 70 set manual_pitch_rate = 70 set manual_yaw_rate = 70 set fpv_mix_degrees = 0 # battery_profile battery_profile 1 set bat_cells = 0 set vbat_cell_detect_voltage = 430 set vbat_max_cell_voltage = 420 set vbat_min_cell_voltage = 280 set vbat_warning_cell_voltage = 290 set battery_capacity = 7000 set battery_capacity_warning = 2100 set battery_capacity_critical = 1400 set battery_capacity_unit = MAH

Jetrell, 2019-09-14:
@JDFPV 
Off hand, the only thing that should make it go into FS, is a very low Link Quality signal. Or the receiver signal is outside the pulse boundary range, which could be caused by an intermittent RX/FC connection.

 Can you please load the `Diff All` files from the CLI. The `DUMP` files you have here are all over the place. Which makes it difficult to look over them.
 Thanks

JDFPV, 2019-09-14:
Dump file

https://drive.google.com/file/d/108oGkaMP15wbvXtFmdxZXytjM4ki2qXw/view?usp=drivesdk

Jetrell, 2019-09-15:
@JDFPV  Thanks. But something that you may not know. File `DUMP` is all the command lines, even the default ones. 
While `Diff All` is only the commands that you have changed.
When saving files and reloading a flight controller with previous settings. It is far better to only use `Diff All` files.  This is because errors can occur when loading all the files.

I have notice a few things that did not add up with his settings. (too many to state here). And too many to surmise what actually happened.
An error may have occurred when loading the files.

I don't know his or your knowledge of the iNav (CLI) commands.  But it is always good to read the CLI wiki and also do thorough ground testing before a first flight.    

One other thing. Did he try to turn off the radio transmitter? That can be useful as a last resort, to insure a FS condition.

JDFPV, 2019-09-15:
Yes jetrell I am aware of the diff command. But my friend that had this issue is out of town working and he sent me the dump files before he left so I could look over them. And he will not be back for several weeks. Yes he even tried turning the radio off as well. You can see the rssi in the video goes to 0 when he does this. And when the radio turns back on the rssi strength goes back up to a normal level. The only control he had from the radio was to the built in camera switcher on the f772 board. He was able to switch between cameras flawlessly. But none of the other functions worked at all. The day before I was with him and he flew this same setup twice with 0 issues at all. So flights 1-4 was fine on this plane. Even flying out at range. Just on this 5th flight trouble popped up.

Jetrell, 2019-09-15:
I understand what you mean.  
This is a time when the log files would be useful. I noticed that you said you don't have them. So it sort of brings us to a dead end.
But I understand how frustrating it is, when an issue occurs and you can't get right to the bottom of it.

As a side note. I never use both FS and RTH mode selection.  
I only use FS. Because flick the switch to enable FS, does the same job as the RTH mode selection. So I can't say if that may have been an edge case issue.

JDFPV, 2019-09-15:
Ok thanks. Could you name some of the other major issues you see in the dump file? Like he changed the looptime to 1000..

Jetrell, 2019-09-15:
I can't see a Looptime change causing this.

`nav_rth_allow_landing = failsafe_only`  
 I always set it to `Never`.  What is odd, is that it should have shut the motor down and circled in, once it was over home... If iNav thought it was truly in FS. 

I am slow to say its a software issue. Because I too had a similar issue some months back. But in my case, the CRSF nano looked up due to a noisy servo and it kept continually rebooting. 
This left iNav thinking that everything was okay. I had no control at all and the plane slowly descended into the ground in AIR mode.

JDFPV, 2019-09-15:
Okay Thanks Again. You have been alot of help. I will tell this pilot to reflash the board. And start from scratch. Not to change or upload anything into the cli. And to make sure the signal between the rx to fc is good..

closed 2019-09-15 by JDFPV

Jetrell, 2019-09-15:
Just one more thing. You stated in the Youtube comments. That it needed tuning. That is true... specifically,  increase the FW_throttle Min and Cruise values. 1200min and 1350cruise, are a bit low. 
Try 1350 for min and 1400 for cruise in autothrottle mode.

JDFPV, 2019-09-15:
I agree as well