iNavFlight/inav #5070: Share you're reliable setup

FlyR9, 2019-09-14:
Hello everyone, im looking to build an inav quadcopter with all of the great features of inav like RTL, PosHOLD, etc 

After choosen my components, learn a lot about inav by the documentation and some youtube videos i would like know if someone have and can post all of his build datas (Full dump, components, pids) based on a racing drone between 120mm/250mm fully operationnal with the base functions reliable.

My goal is not to get all easy without work and research but i dont have the budget to crash and re build, i think this can help more than only me and this can stop to waste our components&money...

Thank you community.

teckel12, 2019-09-15:
@FlyR9 Matek F722SE flight controller and BN-880 GPS. ESCs, motors, battery don't really effect reliability (just performance).

For a frame I'd suggest iFlight. Inexpensive, well-built, and well thought out.

For camera I'd suggest runcam split. Good quality FPV and HD video recording without the need/expense/weight of a GoPro. A little more expensive, but saves money in the long run.

VTx I would spend a little more on. Cheap ones don't perform as well. I've used Matek and TBS VTx and had good luck, others may suggest other good quality VTx.

Not sure if you need advice on transmitters and receivers, but consider your needs. For long range, go FrSky R9 or TBS Crossfire (I use Crossfire with great results). For transmitters, the best (IMHO) is the FrSky Horus X10S. But, its expensive!  Jumper T16 isn't bad, but don't expect the gimbal to last. The Taranis X9D (2019 model) with upgraded gimbal is hard to beat. In that price sweet-spot, high quality (with the upgraded controls) and really does everything you'll ever need (except not in color).

danarrib, 2019-11-06:
Aircraft parts list:
- Sloss 7" Frame
- Emax ECO 2306 1700kv motors
- Gemfan 7042 Propellers
- Racerstar REV35 ESC
- Matek F405-CTR Flight Controller
- AKK Infinite DVR VTX (SmartAudio wired to TX4 pad of the FC)
- Foxeer Lollipop V2 Antenna (MMCX connector)
- Foxeer Predator 4 Mini FPV camera
- FrSky R9MM receiver (SmartPort wired to TX2 pad of FC)
- Beitian BN-880 GPS (wired to TX3 RX3 pads)
- Foldable GPS Mount Holder
- 2x iFlight battery straps

It's a super-lightweight drone for endurance and long range flight. Frame is not very durable tho (so don't crash!). It can be powered by pretty much any 4S Lipo from 1000mAh to 6000mAh or 18650 Li-Ion packs from 3000mAh (4S1P) to 10500mAh (4S3P)

[image]
[image]

INAV 2.2 diff:
```
# diff all

# version
# INAV/MATEKF405 2.2.1 Jul  3 2019 / 22:31:06 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# reset configuration to default settings
defaults noreboot

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# feature
feature SOFTSERIAL
feature GPS
feature PWM_OUTPUT_ENABLE

# serial
serial 2 2 115200 115200 0 115200
serial 4 2048 115200 115200 0 115200
serial 30 32 115200 115200 0 115200

# aux
aux 0 0 0 1300 2100
aux 1 1 2 1300 2100
aux 2 9 1 1300 1700
aux 3 8 2 1700 2100
aux 4 19 3 1700 2100
aux 5 11 1 1700 2100
aux 6 32 5 1300 1700
aux 7 33 5 1700 2100

# osd_layout
osd_layout 0 0 1 2 V
osd_layout 0 1 12 0 H
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 1 3 V
osd_layout 0 9 1 2 H
osd_layout 0 11 25 9 V
osd_layout 0 12 1 9 V
osd_layout 0 13 25 6 H
osd_layout 0 14 1 1 V
osd_layout 0 15 25 2 V
osd_layout 0 20 19 0 H
osd_layout 0 21 1 0 H
osd_layout 0 22 15 2 V
osd_layout 0 23 24 3 V
osd_layout 0 26 25 7 V
osd_layout 0 28 23 1 V
osd_layout 0 30 1 11 V
osd_layout 0 31 5 1 V
osd_layout 0 32 1 10 V
osd_layout 0 33 24 10 V
osd_layout 0 34 11 1 V
osd_layout 0 35 1 8 V
osd_layout 0 37 1 7 H
osd_layout 0 38 1 8 H
osd_layout 0 40 24 4 V
osd_layout 0 41 25 5 V
osd_layout 0 85 25 6 V
osd_layout 0 87 24 8 V
osd_layout 0 97 1 0 V
osd_layout 1 0 1 1 V
osd_layout 1 1 1 10 H
osd_layout 1 2 0 0 V
osd_layout 1 11 25 10 V
osd_layout 1 14 15 0 H
osd_layout 1 15 25 2 V
osd_layout 1 23 24 3 V
osd_layout 1 30 1 11 V
osd_layout 1 31 20 0 H
osd_layout 1 32 1 10 V
osd_layout 1 38 25 0 V
osd_layout 1 40 24 1 V
osd_layout 1 41 1 2 H
osd_layout 1 97 1 0 V

# master
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_stage2_lowpass_hz = 175
set acc_hardware = MPU6500
set acczero_x = -9
set acczero_y = -10
set acczero_z = -7
set accgain_x = 4093
set accgain_y = 4110
set accgain_z = 4115
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 6
set magzero_y = -82
set magzero_z = 9
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 16
set min_throttle = 1050
set max_throttle = 2000
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = RTH
set failsafe_mission = OFF
set current_meter_scale = 240
set model_preview_type = 3
set small_angle = 180
set switch_disarm_delay = 100
set gps_provider = UBLOX7
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set nav_extra_arming_safety = OFF
set nav_auto_speed = 1500
set nav_manual_speed = 1500
set nav_rth_allow_landing = FS_ONLY
set nav_rth_alt_mode = AT_LEAST_LINEAR_DESCENT
set nav_rth_altitude = 5000
set nav_mc_bank_angle = 40
set nav_mc_hover_thr = 1450
set osd_video_system = NTSC
set osd_camera_uptilt = 20
set osd_camera_fov_h = 145
set osd_camera_fov_v = 90
set osd_hud_homepoint = ON
set osd_failsafe_switch_layout = ON
set vtx_band = 5
set vtx_power = 3
set vtx_freq = 5658

# profile
profile 1

set mc_p_pitch = 45
set mc_i_pitch = 35
set mc_d_pitch = 24
set mc_i_roll = 35
set mc_d_roll = 24
set max_angle_inclination_rll = 450
set max_angle_inclination_pit = 450
set dterm_lpf_hz = 75
set yaw_lpf_hz = 0
set rc_yaw_expo = 70
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 50

# save configuration
save
```

FlyR9, 2019-11-06:
Thank you so much for the time you spend to help me this is the perfect setup that i was looking for but it was a long time ago, i give up my inav project for whooping.. however i keep this in mind, i hope be able to try something like that in the future. 

Thank you again!

closed 2019-12-24 by DzikuVx