iNavFlight/inav #5071: iNav 2.2.1 Failsafe Lockout

mrcottonmouth, 2019-09-15:
## Current Behavior
I use RTH after launch to put my goggles on. The plane went to failsafe while in RTH. I could not regain control. It did not do the auto landing. It never went too low on the RSSI, and retained camera controls (using the Matek f722's camera switcher), so I know it was receiving. Ended in a crash.

## Steps to Reproduce
1. Launch plane
2. RTH
3. Failsafe occurred (Not sure why)
4. Can not get failsafe to end and give me control.

## Expected behavior
I expected to move the sticks to regain control

## Suggested solution(s)
Not sure

## Additional context
This is a Sonicmodell AR Wing. I did not have time to tune the aircraft. This was the 5th flight. I do not have blackbox logs, sorry. I have a video: https://youtu.be/AX3EzFXBvL4
and my friend's video: https://youtu.be/0kUwyVIXAk4
Also videos on my channel of it flying fine before.
---

- FC Board name and vendor: Matek F722-WING
- INAV version string: 2.2.1

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/MATEKF722SE 2.2.1 Jul  3 2019 / 22:44:40 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 0 0 0 0 0 0 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
[... 1121 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 30
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 70
set manual_pitch_rate = 70
set manual_yaw_rate = 70
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 290
set battery_capacity = 7000
set battery_capacity_warning = 2100
set battery_capacity_critical = 1400
set battery_capacity_unit = MAH

Ahblair, 2019-11-10:
I encountered the same issue, but eventually regained control after restarting my Crossfire and fumbling with the switches.  My guess is Crossfire enters into mode 1 (300% LQ Scaling) and starts sending regular pulses again.  Maybe there's a threshold built into iNav?

MaxMihawk, 2020-01-14:
I switched from RTH into FAILSAFE (which is set to RTH too). This ends in a horrible uncontrollable flying quad (the compass seems to have an error). I can´t take control back, I even can´t DISARM. It ends in a crash... noone was injured but that was just luck. How can a disarmed quad flying arround?!? That´s fucking dangerous.

closed 2022-06-28 by DzikuVx