iNavFlight/inav #5072: Matek F722 Mini - High Current Draw under USB power

Cro2230, 2019-09-16:
## Current Behavior
Hi, I just flashed a new F722 mini with the latest firmware. 

Under USB power, and no wiring, the Current Draw is showing a steady 33 to 34 amps. 

I'm not sure if this is hardware related or software, or requires wiring and calibration (but would like to confirm before starting to solder everything). I have tested by flashing previous 2 versions, as well as F722-SE and same behaviour. Changing configurations and settings did not change value.

Flashing to Betaflight the current draw there is showing as zero.

[image]

Thanks.

## Steps to Reproduce

1. Flash firmware for F722 Mini
2. Plug in USB
3. Check in Configurator Current Draw value
4.

## Expected behavior

With USB, and no wiring to ADC sensor, I would expect a value of zero.

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: INAV/MATEKF722MINI 2.2.1 Jul  3 2019 / 22:46:46
- INAV version string: GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

DzikuVx, 2019-09-16:
This is not something what firmware can fix. Probable cause is that when on USB power, ADC input of current sensor is not grounded and pin "hangs loose".

closed 2019-09-16 by DzikuVx

teckel12, 2019-09-17:
@DzikuVx You are correct. This is quite common and something the firmware cannot control/correct.