iNavFlight/inav #5075: iNAV GPS page not refresh now , but it refresh last month .(see screen copy)

caosix, 2019-09-18:
Help me please . . . (My English is limited . )
Now my  copter can loiter normally with GPS .

But  It  can't  Navigat .  The  GPS page  has no
data refresh ,  Maybe  it need CLI command ??

Please  see  the  screen copy picture . . Thank you ! 

(I am the first time in github , how can I upload a picture like a forum  ? )

see screen copy by  web page please here :

http://bbs.5imx.com/forum.php?mod=viewthread&tid=1548415&extra=page%3D1

Thank  you  very  much .

digitalentity, 2019-09-18:
This means that GPS doesn't have a fix. Could be a damaged GPS module.

caosix, 2019-09-19:
Sorry  sir  it's  not  the  reason (digitalentity commented 12 hours ago)
I remember very  clear , even if GPS was not  locked , there would still be 
dada  packet  refresh  on  that  page .

caosix, 2019-09-22:
Sep 22 ,
I validated that only when the USB cable is connected directly, there will be data refresh.
I use APM Data Link , there is no data refresh.

DzikuVx, 2019-09-23:
Apparently the APM Data Link is messing something with GPS MSP frames as in the version you are suing it's not a transparent serial connection. I suggest trying different firmware versions for APM Data Link and us generic firmware, not MSP optimized

closed 2019-09-23 by DzikuVx