iNavFlight/inav #5076: Avoid negative current in mAh Calc

FPVZaphod, 2019-09-18:
## Current Behavior
With my Omnibus F4 Pro I have issues to calibrate the current sensor for low and also high currents. Keeping "Offset" at 0, either high currents are correct, or low ones. As an example, I can calibrate the Board for 20 Amps, but then measured 0.6A are shown as 2 Amps in OSD. Long gliding sessions would make the mAh counter very wrong. 
Using the the Offset-Value (60 in my case) makes it possibile to calibrate the current in armed state (VTX running on 500mW) accurate (which is 0.6A), and also higher Current at 20A. 
But doing so, disarming the plane (VTX 25mW) leads to the result, that measured 0.23A are shown in the OSD as -1.6A .  In disarmed state, the mAh counter counts upwards. 
If the current calculation logic would simply ignore negative current, the issue would be solved. 

## Steps to Reproduce
Use Scale 350 and Offset 60 on an Omnibus F4 Pro. Then the OSD is very accurate from 0.6 to 20A, but shows a negative Current of 1.6A while the measured one is 0.23A
With Offset 0 it is not possible at all, to have proper readings for 0.6 to 20A.  

## Expected behavior
Ignore negative current readings in the OSD and the mAh calculation. 

## Suggested solution(s)
Negative current are shown as 0.
Maybe with the option to let the User decide (GUI Switch "Show negative Current". 

- FC Board name and vendor: Omnibus F4 Pro (Genuine)
- INAV version string: 
# INAV/OMNIBUSF4PRO 2.2.1 Jul  3 2019 / 22:07:26 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

marbalon, 2019-09-21:
I know fix is already implemented but I think it should stay as it was. Why ? Two reasons:
- For bidirectional current sensors it is useful to see negative values when you calibrate the zero offset , but If I understand correctly GUI still shows negative values?
- some people try to use solar panels (not so many, I know) and in this situation negative current is useful also.

So I think the best if this is an option not as a forced fix for all setups.

digitalentity, 2019-09-21:
@marbalon fair point. What should we do for non-linear sensors that behave correctly at high amps (in flight) but misbehave and show negative values when idling?

marbalon, 2019-09-21:
I know iNav have already a lot of option in CLI but maybe another flag ?
clamp_current_enabled = true/false?

FPVZaphod, 2019-09-23:
Yeah, the liked issue covers it pretty well I think - at least if we just ignore small currents. 
Another possibilty would be to enter a static current for disarmed state in the configurator. So it could ignore the current meter while disarmed (it's anyways a quite static situation), and then use the current sensor calibrated from armed+idle_throttle to full throttle.
Not sure if there is a reliable bi directional current sensor anyways?

marbalon, 2019-09-23:
I'm using ACS-758LCB-100BP and sometimes 50A version in my planes, and I don't have a problem with a small current. This sensor has really good precision. In my glider, I can spend few our in the air without the motor and calculation are still realy good (+-5%)

FPVZaphod, 2019-09-23:
Yes, Mauch Electronics sells them ready to use on a PCB. But sadly there is no FC with integrated Hall Effect current sensor, they used to be all with a shunt. I would like to avoid this additional part. It seems it's time to design a 38x38 PDB with such a current sensor to at least be able to stack it.

PR #5079 (2019-09-23): [VBAT] Clamp current to positive values
Negative current doesn't make much sense, but it's possible to get negative values on some badly designed sensors (some sensors are non-linear at low currents).

This will fix #5076
closed 2023-03-06 by DzikuVx