iNavFlight/inav #5081: In 3d cruise mode sometimes has a strange behavior and does not maintain the height and does strong variations

tiriad, 2019-09-21:
## Current Behavior
The **3D Cruise Mode** does not maintain the altitude, sometimes performs strong accelerations and decelerations with many delays in reading the altitude.
## Steps to Reproduce
Activate 3d cruise mode
https://youtu.be/egjS9h-lCm8
## Expected behavior
As in this video of another flight
https://youtu.be/bsufwDh-pP0
## Logs
I'm sorry i tried to make blackbox but i didn't get it to record anything on my S800 wing
## Additional context
diff https://pastebin.com/czDLEMty
dump https://pastebin.com/rD1Kkjep

## Version
INAV/OMNIBUSF4V3 2.2.1 Jul  3 2019 / 22:13:58 (a6d847482)
GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

#4878
#5053

Jetrell, 2019-09-22:
@tiriad Can I assume that the two video/flights you posted are the same plane?  And something changed between flights?

Have you tried to re-calibrate the acellerometer?
Has the flight control board moved at all? That could be causing the IMU to think the plane is not level, with respect to the wings angle of incidence.

tiriad, 2019-09-22:
> @tiriad Can I assume that the two video/flights you posted are the same plane? And something changed between flights?
>
> Have you tried to re-calibrate the acellerometer?
> Has the flight control board [...]

Same plane same config, same day, the flight that goes well is earlier and the flight that goes badly is later. I have already noticed this behavior on other occasions but not always and that's why I paid attention to it.  The angle mode works without problems so it doesn't seem like a matter of orientation or calibration.

tiriad, 2019-12-05:
In the last version 2.3.0 I made a flight and there was no problem, I left the pids by default, higher nav_pos_z pid, also  increased Pitch2Throttle that was lower than default and incresed Cruise Throtle a little

closed 2019-12-05 by tiriad