iNavFlight/inav #5083: Can't setup servo gimbal on S5/S6 output

DangerD256, 2019-09-22:
Added two servos in mixer tab:
[image]
But as you can see Servo 0 replaced Motor 1

[image]
Here output is also S1 and no output for second servo.

I need to put servo gimbal to S5/S6 output, how can i do that?

DzikuVx, 2019-09-23:
The board you are using does not support 4 motors and 2 servos configuration. Not every output can be used as servo/motor. Servo 2 was not assigned since board has no valid output for it and servo 1 can not be used together with first motor.
Since you did not reported board name, we can not advise further

closed 2019-09-23 by DzikuVx

DangerD256, 2019-09-23:
Matek F405 CTR, previous firmware was Arducopter and everything worked well with S5/S6
Do i need to open request again as current is already closed?

DzikuVx, 2019-09-23:
INAV supports only 1 servo in multirotor mode on this board by default due to timer conflicts with other functions

DangerD256, 2019-09-23:
> INAV supports only 1 servo in multirotor mode on this board by default due to timer conflicts with other functions

I don't need WS2812, i need gimbal)

ysoldak, 2020-01-28:
I don't know if I'm doing something wrong, but there is a simple fix to have a servo on S5 for this board.

See https://github.com/ysoldak/inav/commit/2094e9870bf68eb8bc9a509fe6c5057b156b249f

Keep in mind though, this feature seem like was explicitly removed from iNav by request from Matek themselves ( see https://github.com/iNavFlight/inav/issues/3223 ), for whatever reason. So this can screw up timers or whatever low level stuff they have there. Works for retractable landing gears on my foamy though...

DangerD256, 2020-02-06:
> See [ysoldak@2094e98](https://github.com/ysoldak/inav/commit/2094e9870bf68eb8bc9a509fe6c5057b156b249f)

Made this change, compiled target SERVO_S6, but S5 still not shown in outputs...

ysoldak, 2020-02-06:
Don't know, mate, it did the trick for me.... Can send you the binary to try, perhaps. PM me.

DangerD256, 2020-02-06:
> Don't know, mate, it did the trick for me.... Can send you the binary to try, perhaps. PM me.

Yeap, please share binary