iNavFlight/inav #5084: Matek F405 CTR Servo gimbal (add servo S5 output)

DangerD256, 2019-09-23:
[image]
Currently only S6 is supported as servo output, is there any way to enable S5?

Prevously i used Arducopter on same Matek F405 CTR and gimbal worked well using S5/S6 outputs...

DangerD256, 2019-09-23:
>I2C requires that the WS2812 led strip is moved to S5, thus WS2812 is not usable on hexcopter and similar.

I don't need WS2812, i need gimbal)

Pete-Slater, 2019-09-26:
I too would like this. Need manual gimble pitch and also output to raise landing skids

DangerD256, 2019-10-01:
> I too would like this. Need manual gimbal pitch and also output to raise landing skids

Use arducopter firmware, works fine, flies well, 2axis servo gimbal works fine

DangerD1024, 2020-03-19:
The problem is that servo S6 also doesn't work...

closed 2022-05-07 by DzikuVx