iNavFlight/inav #5086: How to adjust the Radio PWM input values in Inav (Herelink - Radio)

GanGan7, 2019-09-25:
Hello All,

I am using Herelink radio with Matek F722 FC and iNav 2.2.1, and my problem is that I cannot arm iNav because the PWM range for Herelink Radio is from 1107 to 2009. iNav requires the starting value to be 1000, other wise it will not arm. Is there a way to adjust the input PWM values of the radio in the iNav app?

Thank You.

stronnag, 2019-09-25:
Try the fine documentation, in particular:
https://github.com/iNavFlight/inav/blob/master/docs/Rx.md#receiver-channel-range-configuration

closed 2019-09-25 by DzikuVx

GanGan7, 2019-09-26:
Did not solve my issue, my radio is not listed in the fine documentation. The topic has been closed without providing any solution to the problem.

GanGan7, 2019-09-27:
@DzikuVx  very sorry I am new to INAV and overlooked the instructions thank you very much for pointing it out to me. Sorry for the delay in late reply I had to rebuild my craft and I was able to arm it by following your instructions once more my sincere apology.

I need help in setting up the telemetry via Mavlink, because Herelink keeps video source separately and I will start a new topic to address this issue. Could you please help me with this one too?