iNavFlight/inav #5087: DShot Issue

namirda, 2019-09-26:
Hi - noob here trying to build my first quad.

I have a MambaF7 board and companion 4in1 esc which run Betaflight 4.1 without problem. 

However, I would like to use this quad with INav and there is no supported target for this board. I have therefore made a simple translation of the Betaflight target and everything appears to work well provided I select one of the analog esc protocols.

However, when I try to use DShot (any variant) in my newly minted  INav firmware then motor4 does not spin up. 

Motor 4 also does not spin up using passthrough to BLHelsuite32 if the FC is running INav but it does if the FC is running Betaflight.

These problems arise on the bench with no active VTX or Transmitter and so the problem is unlikely to be due to noise or wiring issues. 

I understand that there are some reservations about DShot in the INav community and I am perfectly happy to continue to use Multishot with INav. However I also understand that DShot300 should work in this situation and I would like to understand what is going wrong. Can anybody suggest any debugging steps I can do to get to the bottom of this? What might be the problem? 

Thanks

Mateyhv, 2019-09-27:
I would suggest you to get a 4.0x target just in case you have translated something related to the new bidirectional dshot that is present in BF4.1 that Inav don't know that even exists.. The problem you have sound quite similar to certain issues that have appeared on some targets in BF 4.1RCx.

namirda, 2019-09-27:
@Mateyhv - Thanks. The MambaF722 is a relatively new board and was introduced for the first time in BF4.0. As far as I can see, there have been no updates to the target files since then. 

It's true that there seems to be an issue for some boards with bidirectional DShot in BF4.1 in which Motor 4 does not spin up - but I don't see how that can be the cause of my problem since I am using blheli32 v32.6 with INav?

teckel12, 2019-09-28:
@namirda Just my opinion, use multishot as it's as fast as DSHOT600 and as reliable as DSHOT 300. And now that you reported the problem, maybe someone with your board will get DSHOT working down the road. Then, you can switch to a slower or less reliable protocol then.

namirda, 2019-12-01:
Problem turned out to be a bug in my target code. See https://github.com/iNavFlight/inav/issues/5189#issuecomment-559941727

closed 2019-12-01 by namirda