iNavFlight/inav #5093: Mamba f405 Mk2 flight controller

hcmak47, 2019-09-27:
Hi 
I'm begging to build my 1st quadcopter 
I have this configuration 
Mamba f405 mk2 flights controller tower  with 40a ecs 
Motor race star b2205 2300kv 
4s batterie 6000mah 
My expectations i want to build a smart drone with 4g conexion.  Is it possible to wire my flight controller with raspberry or any other companion computer 
Thank you

teckel12, 2019-09-28:
@hcmak47 Just to be sure I'm understanding you correctly. You want to communicate with your done via a 4G cellular connection and control it with a Raspberry Pi or other computer?

Were you planning on developing a cellular receiver interfacing with SBUS or a custom protocol? Or were you planning on using hardware currently available?

I'm not aware of any hardware that will do what you're looking for. But if you know of the hardware please share. If you were planning on developing the hardware/software it seems possible with the right skill set.

hcmak47, 2019-09-30:
@teckel12
Yes i want communicate with drone  over 4g connection 
I saw a video for YouTubers using #Andruav and linking a smartphone to flight controller  to flight with it 
And an other smartphone as ground station tu get unlimited  telemetry range 
He said he can controls the drone like that

teckel12, 2019-10-01:
@hcmak47 Let us know how your research goes into this.  I'm not aware of an off-the-shelf solution like this for INAV.  Sounds custom, or not INAV at all.

hcmak47, 2019-10-03:
@teckel12 
try to take a look at this website https://andruav.com/

HefnySco, 2019-10-15:
Hey I am andruav Owner .. yes that is me.
I would like to help please let me know how can Andruav helps you better.

hcmak47, 2019-10-15:
@HefnySco
Thank you for your help 
I contacted you few days ago 
You told me Andruav now support mavlink 2
But i have STM32 f4 flight. I dont. Know  if mavlink 2 with otg telemetey

HefnySco, 2019-10-15:
oic ... yes for the MAVLINK library I use is dronekit for android, and it now supports Mavlink2 and not switchable that is the source of the issue.
if I can make it read one or two in parallel that will solve the issue.
ok I will check that again, and if you can help pls do.

hcmak47, 2019-10-15:
@HefnySco
Im not an expert in this Field 
I was watching your video and it inspired me to make a drone with this app
But it didn't work with me i think because mavlink version
Or my flight controller 
Or maybe telemetey i use 
That's why i try to know the solution of this issue
And thank you to take time to respond me

HefnySco, 2019-12-14:
Hi,
Please download latest Andruav version 3.9.91 it supports version 1 & 2 of Mavlink now.

closed 2020-02-19 by DzikuVx