iNavFlight/inav #5095: Inav some times forgets to add a field in the blackbox .txt

laurentopia, 2019-09-27:
causing inav BB explorer and pidexplorer to freak out (betaflight log viewer has no problem with it)
[image]

the attached file is an example where the bb cannot be read properly by inav bb explorer
the subsequence arms produced good files so it's a transient problem maybe?

[LOG00001.zip](https://github.com/iNavFlight/inav/files/3664658/LOG00001.zip)

[image]

metametaclass, 2019-09-27:
[image]
Difference between bad and good log files

stronnag, 2019-09-28:
Possibly excessive loop time / log rate (500, 1/1). Hardware can't keep up. 
Can you log consistently at lower rates?

DzikuVx, 2019-09-30:
Too fast loggings is the most probable reason

closed 2019-09-30 by DzikuVx

laurentopia, 2019-10-09:
The SD does 12 MB/s write, what do you recommend log% for this loop time?

[image]