iNavFlight/inav #5099: Servo Pass-through Causing Servo Craziness

atheimer, 2019-09-28:
I recently upgraded my airplane's 411-Wing Inav to 2.2.1.  I did a fresh flash and re-configured from scratch.

When I boot up inav all my servos are driven full power for 10s or so until *the boot finishes*?
This has already broken one of my servos so when this happens the servo just keeps moving in circles because it doesn't have a physical obstruction.

I have been able to reproduce this event several ways:

1. If I plug the battery in with the TX off
2. If I add a Channel Forward rule to the mixer (for flaps channel in my case) and click Save and Reboot

I have tested the following:
- 2 different RX's no difference
- TX failsafe is set to No Pulses (D16 sbus)

Video of Servo Behavior: https://www.youtube.com/watch?v=ibi5ztvWzGY

In the video you will see:
00:06 A power up with the TX on and NO forward rule = good power up or what I would expect.
00:42 A "save and reboot" after adding a Channel Forward rule (servo 6 > RC Channel 7) = crazy servos

I have not touched the Failsafe features at this point so they are default.

# version
# INAV/MATEKF411 2.2.1 Jul  3 2019 / 22:50:30 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

Full CLI dump attached 
[INAV_cli_Tundra_20190928_122944.txt](https://github.com/iNavFlight/inav/files/3665810/INAV_cli_Tundra_20190928_122944.txt)

atheimer, 2019-09-29:
More testing, tried flashing Matek411 2.2.1, 2.2.0, 2.1.0  that same issue exists that servos do some bad things without signal.

I went back to 2.0 no issues at all.

AND.... did a dump diff, reflashed 2.2.1 imported the diff  plugged battery in and
crazy servos... 

So... either a problem with my Matek411 (i don't have this problem on my AR WING running a 411 and 2.2.1) or something in the config? Dunno... there is not other info that I can find on this topic.

Jetrell, 2019-09-30:
@atheimer There was an addition in 2.1 that increased servo mixing weight for gimbal use. 

I experienced it having the same effect with Analog servo's. But only if I also had a Digital servo connected all with the Analog servos.   (I was using an F411 as well)

closed 2020-02-19 by DzikuVx