iNavFlight/inav #5103: It would make sense to reverse one servo for flying wing mixes

tigert, 2019-09-29:
## Current Behavior
Currently no servo reversion is set up automatically, as this is highly dependent on the aircraft model construction.

## Desired Behavior
However, most of flying wings with elevons have the servos on the top of the wing, mirrored to each other in their orientation.

This means one of those servos needs to be reversed in order to make the elevons work correctly.

## Suggested Solution

Maybe the predefined example configurations ("S800 flying wing" etc) could have the servo reversals defined, so the pnp/kits would work out of the box as far as servos are concerned.

## Who does this impact? Who is this for?

It is a very common case for newcomers to get started with something like the AR wing or other fpv wing kit. It would be good to have the predefined setups also take the servo orientation into account, and have the correct reversal set up by the preset.

Jetrell, 2019-10-01:
@tigert  The other issue here, is not all servos rotate in the same direction by default.  Some will go clockwise normally and others anti-clockwise.

DzikuVx, 2019-10-01:
Also, which servo to rotate? Left or right one? All of that is highly dependent on the build itself and hardware used and there are no defaults that will work for everybody with 100% success ratio.
Such changes were proposed before too. I do not think we will implement it as it will not fix all problems. Thank you very much for submission, but I will close it tho

closed 2019-10-01 by DzikuVx

tigert, 2019-10-01:
Good points. 

I think what could be done to help the actual issue is the model profiles - at least for popular pnp wings and other kits. 

I assume the presets can specify any configuration options? I will try with my mini ar pnp and see what I can do. At least that one has preinstalled servos with that are supplied with the kit.