iNavFlight/inav #5105: F405-Wing: Please increase nuber of available mixer rules

SquashSquishSquaw, 2019-09-29:
## Current Behavior
Currently, the mixer for the F405-Wing provides a maximum of 16 mixing rules.

## Desired Behavior
The F405-Wing is a flight controller for fixed wing airplanes, which often have a complex mixer setup. For a feature-rich full house glider setup, even 32 rules aren't sufficient. More mixer rules would be necessary.

## Suggested Solution
Increase the number of available mixer rules to 48 or 64, but **only if** the available memory on the F405-Wing allows to do so without risk.
The relevant parameters seem to be in inav/src/main/flight/servos.h :
#define MAX_SUPPORTED_SERVOS 8
#define MAX_SERVO_RULES (2 * MAX_SUPPORTED_SERVOS)

## Who does this impact? Who is this for?
The feature request addresses the needs of airplane users using the Matek F405-Wing flight controller.

DzikuVx, 2019-10-25:
INAV 2.3 will have support for 16 servos and 32 servo mixer rules.

closed 2019-10-25 by DzikuVx

SquashSquishSquaw, 2019-10-25:
What speaks against increasing this to 48 mixer rules, please?

And, is there no way to do a dynamic memory allocation, which only consumes memory when rules are specified, but no memory when they aren't?