iNavFlight/inav #5106: DJI FPV cannot change PID in-goggles

laurentopia, 2019-09-29:
2.2.1 firmware on a matek f405-ctr
repro:
enter PID turning in the DJI-FPV goggles, the proper values show up
select one value and change it: the value changes but is rapidly reset to the value that was set in the configurator
the DJI air unit is connected via MSP to tx1/rx1
PID tuning works in betaflight

[image]
# version
# INAV/MATEKF405 2.2.1 Jul  3 2019 / 22:31:06 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

https://pastebin.com/50hKLvLy

digitalentity, 2019-09-30:
It's likely that DJI is sending a `SET_PID` message in a format that's incompatible with INAV. Unfortunately non of the developers has the DJI hardware to debug.

Maybe somebody from @DJI-SDK can chime in on who at DJI side can help.

digitalentity, 2019-09-30:
Generally, I see this as a 3-rd party device compatibility issue. Two options here:

1. INAV will correctly accept new PIDS if DJI air unit will send a properly formatted `SET_PID` message. This is the issue DJI should fix in their firmware.
2. Alternatively we can add a work-around on INAV side to accept whatever DJI air unit is sending. In this case we would expect DJI (or other interested parties) to follow the [hardware policy](https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md) and provide us with devices to debug and test.

laurentopia, 2019-10-09:
10 days and still no response from DJI so it's up to us.
I have a set of goggles and the inav build environment works so I could do it but I need some help sifting through the MSP code:

1. MSP_SET_PID is the same as betaflight apart from the error catching, _ADVANCED however is very different but doesn't seem to set PID values, only F and a few exotic things. No PID in ADVANCED?
2. Is there a SET_PID which also sets F? I'm tempted to say that the DJI send a SET_PID followed by a SET_ADVANCED but it doesn't explain why SET_PID is ignored by inav...
3. Is there a flag to dump whatever MSP receives to the log file?
4. MSP_PID seems to be what the goggles send after each MSP_SET_PID (or _ADVANCED) to refresh its display, are each command self contained or do they require something like an OPEN and CLOSE to bracket them, or are they batched in any way?

thanks.

SystemsGuy, 2019-12-04:
It would seem to be in our best interest to simply align with BetaFlight until such time as DJI acknowledges that iNav is a thing.  

@laurentopia I've never looked at the MSP code, but I'll try and carve some time out this weekend.

closed 2023-10-27 by DzikuVx