iNavFlight/inav #5110: Target request: FLYWOOF722DUAL

lossUAVes, 2019-09-30:
New FC with baro and 20x20 mouting holes. Very interesting.
[image]
https://flywoo.net/collections/frontpage/products/goku-fc-f722-mini-dual-flight-controller-built-in-osd-5v-9v-2a-bec-icm20689-mpu6000-for-rc-drone-1

DzikuVx, 2019-10-01:
@lossUAVes please see https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

lossUAVes, 2019-10-01:
I'm sorry. I was just trying to help.

DzikuVx, 2019-10-01:
@lossUAVes no need to apologize. I'm just informing what is the current INAV policy towards new targets

lossUAVes, 2019-10-01:
Thank you. I did not know it. I thought you could migrate from betaflight.

echma, 2020-06-07:
Hi. I'm sorry to open this again but is there no news? I was just looking through INAV 2.5 and this board is sadly not supported. I only found this build: inav_2.3.0_FLYWOOF7DUAL.hex

digitalentity, 2020-06-08:
@echma please see https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

TL;DR: we need samples and specs from the manufacturer to be able to add this target. This is done to avoid having a million of scarcely used targets.

closed 2022-05-12 by DzikuVx