iNavFlight/inav #5113: Can I fly

LawrenceLDJr, 2019-10-02:
## Current Behavior

## Steps to Reproduce
1.Install INav 4.2.1
2.New Airbot F4 V6, Tyhpoon 30A 4 in 1 ESC
3.Choose synchronize looptime with gyroscope 
4.Every reboot of INav deselects this option

## Expected behavior
My X220S has this option enabled and flies great. I enable this option with this quad and it switches off on every reboot. I have ESCs calibrated, Multishot 2k, gyro lpf 256hz, flight control looptime 2k, same settings as the X220S. Synchronize looptime won't stay enabled. Gyro, accel, mag, baro, gps all blue, all calibrated.

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

DzikuVx, 2019-10-03:
`Every reboot of INav deselects this option`
That mean option is not supported by hardware. It's still safe to fly

closed 2019-10-03 by DzikuVx

LawrenceLDJr, 2019-10-03:
Thx Pawel... gonna try in a few minutes...