iNavFlight/inav #5114: Option to chop off the first four digits of the plus code, since they're optional

skorokithakis, 2019-10-02:
## Current Behavior
Currently, INAV shows more than 10 digits of a plus code. The first four digits are optional and can be omitted by specifying a general location (like the city you're flying near).

## Desired Behavior

Adding an option to not show the first four digits would make the plus code display much shorter and more convenient.

## Who does this impact? Who is this for?
Anyone who uses plus codes could benefit from this.

LawrenceLDJr, 2019-10-03:
You can choose the amount of digits you desire...!

skorokithakis, 2019-10-03:
Yes but it doesn't go lower than ten and you can't cut off the first four, so you lose accuracy.

avs2, 2020-05-14:
I second this request. The plus code could be significantly shortened. Added benefit is that your location is at least somewhat obscured when posting DVR footage online.

avs2, 2020-06-17:
Just commenting again in the hope that one of the devs see it. I think this should be very easy to implement, but I don't have any experience with this programming language so I cannot really help myself.

avsaase, 2021-01-04:
I managed to figure out how to do this. See the above linked PR.

PR #6469 (2021-01-08): Option for local plus codes on osd
This PR resolves #5114 by implementing a CLI command `osd_plus_code_type` with options `GLOBAL` (the default) and `LOCAL`, where `LOCAL` chops off the first four digits of the plus code. To find the original location you'll need a reference point like a nearby city or mountain within 0.4 degrees longitude and 0.4 latitude, roughly 40 km.

~~The configurator does not yet understand this option so it doesn't allow placing the shorter plus code against the right edge of the screen. To position the osd element like in this screengrab you need to use CLI command ` [... cut]
closed 2021-01-08 by DzikuVx