iNavFlight/inav #5116: Please implement dynamic weighting in smix

SquashSquishSquaw, 2019-10-04:
## Current Behavior
Currently, the iNav mixer does not allow dynamic weighting of mixer rules.

For example, it is not possible to change the weight of a "stabilized roll" rule during flight.

## Desired Behavior
Please implement dynamic weighting of mixer rules.

## Suggested Solution
The implementation is most likely extremely easy and quick to do:

- Currently, the "Weight (%)" field accepts only integer numbers, usually ranging from [-100..+100] %.
- "RC Channel n" Values as specified in the "Input" field also usually range from [-100..+100] %.
- If the "Weight (%)" field ist made to accept "RC Channel n" as an input, the weight could be changed during flight by an RC channel.

## Who does this impact? Who is this for?
This feature request impacts all those users who would love to use the mixer for a dynamic setup.

Example: Airplane: Both ailerons are being lowered during flight by a changing value of n %. The usual, standard "stabilized roll" acting on the aileron servos therefore needs to be weighted down to (100 - n) %. Currently, this can only be done in a static way by setting n to the maximum possible of all expected values. The suggested fix would allow proper adjustment "on the fly".

DzikuVx, 2019-10-06:
I fail to see a gain of such a function. Then both ailerons are lowered together, they do not affect roll axis, so there is no need to change weight of roll mixes at all.

SquashSquishSquaw, 2019-10-06:
There are lots of applications.

Example 1: A glider with a "Camber" setting for thermal soaring. In order to increase airfoil lift, both ailerons are being lowered by angle A and both flaps are being lowered by a second angle F. For simplicity, let us only look at angle A.

Depending on thermal uplift, angle A needs to be adjusted _during flight_. Let's imagine we start with A = 5 degrees. Both ailerons are now pointing downward by 5 degrees. In order to achieve these 5 degrees, we need to rotate both of the the corresponding servos by e.g. 20 %.

Of course, these 20 % are now missing for the stabilized roll which the aileron servos are usually intended for. Thus, we would decrease "Weight" in the corresponding "Stabilized Roll" mixer rules from 100 % down to (100-20) % = 80 %.

Now, assume that during flight, thermal conditions change in such a way that angle A needs to be increased to 15 degrees downward deflection. Both ailerons now point downward by 15 degrees. Assume furtheron, that in order to achieve 15 degrees of deflection, we need to rotate both aileron servos by e.g. 60 %, respectively. At this point, we would need a decrease of the stabilized roll down to (100 - 60) % = 40 %.

With a static "Weight", we lose lots of agility. All we can do is to look at the maximum deflection MAX which we will ever have (e.g. 20 degrees down, 70 %), and do a permanent weighting of the stabilized roll down to (100 - MAX) %. This is possible of course, but we lose a lot of aileron stabilized roll just because of the very rare cases in which we will use 20 degrees of camber deflection.

With dynamic weighting on the other hand, we can adjust camber deflection from x % to y %, and on the fly readjust aileron stabilized roll to (100 - y) %.
_______
Use case 2: Imagine a "Landing Mode" setting for an airplane in which we use "crow breaking". The further down we deploy the flaps and the further up we lift both ailerons, the less stabilized roll we will have. At the very end, with maximum crow breaking, both flaps will be completely down, and both ailerons will be lifted up by a considerable servo throw.

Again, this concerns only a short period of the flight, because usually we crow break for only very short periods of time. Yet, we must make sure beforehand that servo throws never go beyond 100 %. So again, we would have to decrease aileron stabilized roll by a considerable amount, making the plane much less agile than it _could_ be, with dynamic weighting.

With dynamic weighting, we start with full aileron stabilized roll agility in Landing Mode. The more we brake, the less agile it becomes, but as soon as we release the brakes a little, we immediately regain agility.

I am sure that there are numerous other possible use cases which maybe don't even have to do with flying itself. Why restrict "Weight" to something static, when it's (hopefully) so easy to make it dynamic?

All considerations apply to flap servos as well, of course, when we use more sophisticated setups with e.g. full wing ailerons in which the flaps support the stabilized roll of the ailerons.
_______
Use case 3: Imagine, we want to apply a smooth curve to a complex setup. With the current smix, I believe that this cannot be done.

DzikuVx, 2019-10-07:
As for examples 1 and 2, this can be implemented with current mixer without a need to modify anything. Mixer rules are additive and if you add a rule to deflect a servo 20% down for example, it will not affect the rule that brings stabilized roll and PID/PIFF will compensate.
The only difference between your proposition and current implementation is what will happen is ":stabilized roll" will go about 80% of deflection. In your proposition the total weight would be reduced. In current implementation it would be clipped at 100% of deflection.

Flaps, crow brakes, cabers and so on can be implemented currently without changing the mix weights.

gamaxgbg, 2019-10-15:
I really liked this idea of helping in situations with reduced throw, but i didn't like your solution of dynamic weight. It would be a lot better if it were dynamic end points (or dynamic subtrim). In transmitters, subtrims are used to shift the entire pwm band to one side, including min end point, the middle, and the max end point. 

Usually, the correct way of trimming a new plane is to set a middle point where you think would be the correct for the plane to fly straight and then setting the end points for that surface to have the same degrees of deflection from the chosen middle point for both sides. If the middle point is wrong, you can use trim knobs to set a new mid point, but then you would have more deflection to one side than the other, that is why subtrim is used, to shift the new mid point but also continue with the same deflection for both sides.

For a flaperon, you'd want to do a subtrim for the desired position because then you'd have the same deflection afterwards, and not a limited one. But i think you can achieve that by raising all your end points overall and giving just a percentage in the mixer instead of 1. For example, if your flaperon's mid point is 1500 and end point 1700 and you want to add a FLAP offset of 100 (or a new 1600 middle) and not lose 50% deflection, then you just need to also add 100 to your end point (to 1800) and set the difference in the ROLL mixer (200/300 = 66.6%) and your roll would always move 200.

I think this is what you wanted, right? But this will probably not work if the end points goes further than 850-2150. In this case, something that dynamically reduces the FLAP offset as you input more ROLL would do better. I don't know if it would be that useful, because usually you can raise your throw mechanically if you need more, and situations that use double inputs in a single servo usually don't requires the static input to be that high, like 80% FLAP, continuing the same example. It is almost always an acceptable amount to still let you stay between the 850-2150 range.

SquashSquishSquaw, 2019-10-21:
Dynamic end points are also a great idea, of course. Perhaps they are a little bit more complicated to inplement into smix, but I would love to have them as well.

> Usually, the correct way of trimming a new plane is to set a middle point where you think would be the correct for the plane to fly straight and then setting the end points for that surface to have [...]

I'm trying to understand this. Could you please first explain, with maximum possible verbosity, how you define "flaperon", "flap offset", "middle point" and "end point"?
_______
What I would like to have is a mixer in which I have complete control over what I'm doing. Obviously there is a difference in mathematical terms, between clipping a function on the one hand, and shrinking it, on the other. In some cases I want the former, in other cases I absolutely do want the latter.

Being a newbie with RC programming in general and flight controller programming in special, I am not yet sure what I really need. But I did have large difficulties setting up my glider functions with the things which I had at hand. After a few weeks of trying, I finally managed to get it working with the current mixer, but I still feel that it could be done in a way less awkward way with dynamic mixing.

One problem is what I think you refer to as "situations that use double inputs in a single servo". Ignoring mechanical solutions, I think that currently, there are only two options: Either to rely on iNav not messing up with the clipping, when servo throws calculate to over 100 %, or to do a permanent, hard-coded reduction of one of the two inputs.

In the first case, when clipping is in effect, then I believe that the translation of stick movement to servo function is also getting clipped. For example, when aileron clipping needs to be done at 20 % aileron deflection in order to prevent servo damage: When my aileron (roll) stick on the transmitter has moved 20 % and clipping sets in, then the rest of the 80 available % of the transmitter stick movement will add no further aileron roll deflection, i.e. further stick movement will do nothing at all. On the other hand, with a "shrinking" (and non-clipping) solution, the 20 % would get spread over the full 100 % range of the stick movement, which is something completely different than in the first case. If you want the latter, then currently, you need to hard-code maximum aileron roll deflection to the completment of whatever maximum is expected for the second function on the servo. This is what I had tried to describe in example 1.

Perhaps there is some very clever way to get around this problem which I haven't been able to come up with, yet. Currently, I have chosen the clipping solution, which leaves me with a nervous feeling because I don't know how reliable the clipping algorithm really is.

What saddens me more is that I had to give up on my hopes of being able to implement landing mode with crow breaking in a rather sophisticated way:

1. I wanted to have the flaps go down the first half of their way without aileron addition (except aileron roll of course, but no crow, i.e. just usual flaps function, *but* see 2.).

2. During their complete range of possible movement, I wanted to have the flaps support any aileron roll a little (I learned this to be called "full wing aileron"... is it what you refer to as "flaperon"?). There are two extremes:
2a. Of course, when the flaps are deflected fully downward, only one of the two flaps can move upward in order to roll.
2b. On the other hand, when the flaps aren't deflected at all, i.e. when they are level with the airfoil, then in order to roll, one flap can go to below the airfoil, and the other can go to above (my flaps allow this, since I have set them up to be able to go above the airfoil surface in order to have "wing reflex" as well).
2c. Anywhere in between these two extremes 2a and 2b, I find it desirable not to destroy the usual, standard function of the flaps, which I believe is to slow down the airplane *and* to increase lift at the same time. Thus, for roll movement, I'd like to have one of the flaps go down more than the other goes up. This can indeed be done with the very useful "stabilized roll+" and "stabilized roll-" smix functions.

3. But as you can probably imagine, a smoothing function must be applied in order to avoid mathematical discontinuities when going from 2a to 2b and back, especially at the mid-point of flap movement, when the flaps are half way down and further downward movement starts to bring up both ailerons for crow breaking.

4. When crow breaking sets in after half of maximum downward flap movement, i.e. when both ailerons start to move more and more upward, I want them to behave exactly the opposite as the flaps do. I don't want to destroy the aileron's "crow/spoiler function" when a roll is being engaged. If at all, I want the roll to increase crow breaking a little. Thus, one aileron needs to move more upward than the other does move downward. Again, as in 2a and 2b, there are two extremes:
4a. Both ailerons are level with the airfoil surface. This is the case when both flaps are down half way, and
4b. Both ailerons are deflected upwards to their maximum crow deflection.
Again, care must be taken to avoid any mathematical incontinuities when moving from 4a to 4b, and to get a smooth curve progression.

The smoothing would have been possible to do with dynamic weighting in smix. Without dynamic weighting, even with dynamic end points at hand, I wish anyone who is attempting to do this, a lot of fun. :-)

DzikuVx, 2022-02-23:
This can be now implemented with Programming Framework, no need to develop anything special

closed 2022-02-23 by DzikuVx

dmstan, 2024-04-23:
I would love to use dynamic weight to change the servo seamlessly from a VTOL motor tilt to elerons

sensei-hacker, 2024-04-23:
> To ensure a smooth VTOL transition, incorporating servo dynamic weight would be beneficial. For instance, I require a method to switch the servo from tilting a VTOL motor to controlling elevons/flaperons. Currently, there's no viable way to execute this transition seamlessly. 

Are you thinking in terms of the native INAV VTOL support, or trying to implement your own separate VTOL support?

amstan, 2024-04-23:
> Are you thinking in terms of the native INAV VTOL support, or trying to implement your own separate VTOL support?

More of the latter, but i'm unclear where the differences are. From our novice viewpoint it seems Inav's current VTOL support relies on a craft being able to instantly change modes (as long as there's sufficient speed). But our craft is weirder (a quadcopter where every motor can be tilted to go forward, then the tilt mechanism is used for control in horizontal mode). This discontinuty is hurting us near 45 deg (ex: any roll becomes almost as much yaw and so on).

dmstan, 2024-04-23:
I was thinking in terms of the native INAV VTOL configuration. There is no smooth transition between VTOL and wing-borne mode, especially for a tilted rotor. I'm using a slider to perform transition, and I think as the motors are tilted, more and more the servos should start acting as the fixed wing controls. Transition have a lot of aerodynamic non-linear effects, adding control discontinuities does not help. I prefer having more options to tweak controls and dynamic weight would be very helpful in this regard.

amstan, 2024-04-24:
This is essentially what we're looking for: https://youtu.be/RabFZzRyZo8?t=485

Replace the `transition_fader` variable with an analog RX channel driven by a slider.