iNavFlight/inav #5117: Support for change vtx power at channel value

whoim2, 2019-10-04:
Now change vtx power present on Ajustments, but this need 2 channels for use it. First we active mode "change value", second we changed this value up or down.

More lrs change self power at one channel, 3-pos switch. This channel present on FC.
First position - low level, second - mid level, third - max power (e.g. qczek lrs).
VTX may change self power via smartaudio at this one channel?
Example, from cli set range channel values and vtx power level, vtx_power_channel=8, vtx_power_channel_level1=1000-1299, vtx_power_channel_level2=1300-1599, vtx_power_channel_level4=1600-2000.
?
//Sorry, my english is bad)

DzikuVx, 2019-10-06:
It is possible already using Adjustment and in INAV 2.3 will be possible via Global Functions.

closed 2019-10-06 by DzikuVx

whoim2, 2019-10-06:
In ajustments i not may set power from ONE 3-pisitions switch. Ok, wait 2.3, thanks!