iNavFlight/inav #5118: Copter disarms in the air

Burnst81, 2019-10-05:
## Current Behavior
I was flying without any issues.. no Problems.. About 500 - 700m away from homepoint. The copter disarmed in About 50 meters of heigh. Unfortunately i didnt record the flight in my googles, but i got it on the HD-Video of the Runcam split mini 2 with Sound. I can hear cleary how all the motors were switched off immediately at the same time. The Drone rolled on the back and came done like a Stone. Also there was the flight summary displayed while falling.. so a complete disarm. There was no flyaway from one Motor or something else which lets me guess that there was an Hardware issue. Further more i flew 3 or 4 lipos more after i picked it up, without any issue.. 

This happend at Minute 3-4 of flight time. Before that, i tested RTH in the same flight and it worked fine. I also set RTH as the failsafe procedure. 

The was no "rx-loss" displayed in the Moment of disarming. No Input lags or something else..  I also didn t touch the arm switch. 

Unfotunately the flight has been overwritten in my 16mb Blackbox Memory. 

I try to find out what the reason for this was. i already had an rx-loss with this quad, but it switched in the RTH-Mode like it should without any Problems. Can somebody help me to find out?

Regards

## Suggested Solution
I set the guard time for failsafe stage 2 to 4.. since my last flight.. could that be in context with the issue?

My board is the kakute F7 mini with multishot on a 3 Inch quad.

yo8aiv, 2019-10-09:
wich version of inav. at least post a _diff all_  for the dev's . 
i'm worrying of upgrading to latest due to dshot issues i'm hearing but seems your problem might be different. 
check that your gps works properly. analog dvr would be nice to have. 
i'm interested to see results of this. cheers.

danarrib, 2019-11-06:
> Also there was the flight summary displayed while falling

In the summary, on the last line, there's a line telling the reason of disarm.
Example: 
"DISARMED BY: SWITCH"

What did your summary data display?

[image]

closed 2019-12-24 by DzikuVx