iNavFlight/inav #5119: Sticks starting engines

Revenger72, 2019-10-06:
I’m sorry, I don’t know English (I use Google), so I can’t find the right question or answer.

Why did they remove the ability to start engines with sticks from fresh firmware? Not everyone is used to doing this with a toggle switch, why are other users being infringed? You can also turn off the motors with the toggle switch (which I did by accident yesterday, I almost broke the wing), why did they go “on the occasion” with those who foolishly dropped the devices with sticks?
thank

DzikuVx, 2019-10-06:
Stick arming was removed by purpose and will not be back. With modern radios there is no practical reasons not to arm with switch

closed 2019-10-06 by DzikuVx

Revenger72, 2019-10-06:
Strange attitude towards users. Align all under one. For some reason, there are a lot of hidden parameters in the "Kli" with which you can make trouble, and they are not cut out, and the Stick arming was removed . One more step in disappointment.

DzikuVx, 2019-10-06:
If you want stick arming you can go back to INAV 2.1 or even earlier versions

Revenger72, 2019-10-06:
Yes thank you. On quadrocopter 1.8 firmware and very satisfied. It is a pity that the developers remove the necessary functions (unlike some clearly not needed) at their discretion.