iNavFlight/inav #5120: Mixer servo speed bug?

SquashSquishSquaw, 2019-10-08:
## Current Behavior
In the **smix** **"Speed (10μs/s)"** column, a value for servo speed can be specified for a given mixer rule.

Currently, this seems to work only when for the very first time after reboot, a rule is being executed. On subsequent calls of a mixer rule, the default servo speed seems to get applied, disregarding the user specified value.

## Steps to Reproduce
1. Add a mixer rule, e.g. for a flap servo:
Rule# Servo# Input Weight(%) Speed(10us/s) Active
1 5 "RC Channel 5" 100 10 "Always"
2. Let the servo move, e.g. by an instant change from +100 % to -100 %, and observe servo speed:
On first execution, servo will move very slowly
3. Let the servo move a second time with an instant change from +100 % to -100 %, and observe servo speed:
On second execution, the servo will "hop" instead of moving slowly

## Expected behavior
When specifying a servo speed, the specified value should get applied at every call, instead of only at the very first time.

E.g., flaps usually are supposed to be deployed slowly in order to prevent jerky movements of the plane. This is always the case, not only at the very first time during a flight.

## Suggested solution(s)
Please check if this can be reproduced, and perhaps change the algorithm accordingly, unless it's a "feature not a bug".

## Additional context
I'm running a modified version of iNav on the Matek F405-Wing, with the only code change being an increase of the maximum number of mixer rules from 16 to 48 total (edited by increasing the maximum number of servos by a factor of 3, which also increases the available number of rules). Currently, around 40 mixer rules are active. This does not seem to influence the problem, though.

---
# version
# INAV/MATEKF405SE 2.2.1 Sep 27 2019 / 15:17:43 (af08641f9)
# GCC-8.3.1 20190703 (release) [gcc-8-branch revision 273027]

- FC Board name and vendor:
Matek F405-Wing

- INAV version string:
(I am sorry, I do not know how to get the INAV version string)

DzikuVx, 2019-10-09:
@SquashSquishSquaw please provide CLI `diff` result

SquashSquishSquaw, 2019-10-12:
I've sent you the diff via email.

DzikuVx, 2019-10-13:
The problem is that servo speed is applied on smix level. When you deactivate a rule via LogicCondition that servo is allowed to travel at full speed to mixpoint or any other position determined by other mixer rule. Truth to be told, we might want to change it. I'll make a ticket to implementet servo level speed limit in a future

closed 2019-10-13 by DzikuVx

SquashSquishSquaw, 2019-10-13:
Many thanks for cheking this.

I'm not sure if making servo speed a property of each servo is the best way to go. In my case, I do want flaps to deploy slowly, and at the same time I want mixed in servos like elevator to go slowly just alike. But while the mixer rules for deploying flaps and the connected elevator compensation need to go slow, the standard elevator function, and perhaps also a stabilized roll function on the flaps, need to be fast. Otherwise, the plane would lose a substantial amount of agility.

I can imagine that this is quite difficult to implement, since one servo would be doing two movements simultaneously, a slow one, and a fast one.

OpenUAS, 2019-11-07:
@SquashSquishSquaw if you have those wishes, maybe consider that it is time to start using Paparazzi? Things like <ratelimit var="brake_value" value="$brake_value_nofilt" rate_min="-BRAKE_ENABLED_SLOWDOWN_FACTOR_RETRACT" rate_max="BRAKE_ENABLED_SLOWDOWN_FACTOR_EXTEND"/>  etc. etc.

SquashSquishSquaw, 2019-11-07:
> 
> 
> @SquashSquishSquaw if you have those wishes, maybe consider that it is time to start using Paparazzi? Things like` <ratelimit var="brake_value" value="$brake_value_nofilt" rate_min="-BRAKE_ENABLED_SLOWDOWN_FACTOR_RETRACT rate_max="BRAKE_ENABLED_SLOWDOWN_FACTOR_EXTEND"/>''` etc. etc.

Hi OpenUAS,

many thanks for your suggestion. I've just had a look at the https://wiki.paparazziuav.org page, and it does look interesting. Is there an up to date list of supported hardware? The list at https://wiki.paparazziuav.org/wiki/Category:Autopilots seems a little dusty.

I'm more or less new to flight controller software, so I'm still learning. I've been successful with ArduPlane on small gliders, and Ardu gives me almost everything I want, except that I find the mixer still a bit rudimentary. I hadn't been able to implement all my setup wishes which I had had for my full house glider. For this reason, I decided to give iNav a try, and so far I'm really happy with it.

In fact, I wouldn't really want the iNav developers to spend much time on the "issue" which I've described here. I could imagine that there are quite a lot of people who would also be interested in a solution, but on the other hand, it is indeed possible to do the slowing down of servo movement on the TX software, i.e. OpenTX in my case. Finding a good solution inside iNav would probably waste lots and lots of valuable developer's time, which can be spent much more beneficially elsewhere.

So all in all I'm quite content, but since learning is always fun, I will be happy to have a closer look at Paparazzi, provided that I see a chance for me to implement complicated mixing setups, and provided that I don't need to spend loads of money for proprietary hardware. Links, please! :-)