iNavFlight/inav #5121: OSD problem / bug?

jagwal1, 2019-10-09:
Inav 2.2.1
Matek F722-wing

My apologies if I post this in the wrong section, but I'm not sure at this stage if my problem is related to my flight controller, Inav firmware, or Inav configurator.

At first I was able to setup my OSD perfectly.
Only at startup I was getting the message: "Invalid Font"
So I uploaded a font, then my osd after reboot was filled with small characters (see attachment photo).
I tried different fonts, all the same fault. I tried disconnecting sensors (interference?) and other peripherals, same fault. Different camera, same fault. Re-flashed board, full chip erase, same fault.
Tried Auto,PAL and NTSC setting, same fault.

Then I started with a blank OSD, disabling everything. One by one I started adding features and... bingo ! As soon as I enabled "System Messages", the screen went haywire again.
Even 'System Messages' on it's own created the same issue.
Just disabling 'System Messages' fixed the issue after a reboot.

My questions are;
- Why I could use 'System Messages' in the beginning AND I could setup a full OSD while there was no font loaded ('invalid font' message)
-Why did loading the default font in the font manager not fix the issue, neither did "full chip erase" (delete problematic font?) when I re-flashed. 

Youtube clip of screen: https://youtu.be/baEhU_KPM4w

Thanks in advance!

DzikuVx, 2019-10-09:
@jagwal1 was System Messages added as a last line in the OSD?
Could you please check if the same is happening is System Messages is not one of the last lines?

jagwal1, 2019-10-09:
Thanks for the quick reply !!

I just checked and yes, as soon as I move it up, the problem is solved. Thank you Pawel !
Still I would prefer in certain screens the message on the bottom, could this be fixed in the future?

DzikuVx, 2019-10-09:
Fixed? Probably. We only need someone to spend some time looking for a reason and implementing a fix

DerElch87, 2019-10-09:
I experienced something similar with Inav 2.2.0 and a matek f405 wing.
As soon as certain characters (not all) were moved to the outermost position (left or right) everything was kind of blinking.
I could do it with a restore of the osd dump all the time.
I figured out that it only was a problem with some fields, so I came to a solution with try and error.
When it failed, I rebooted without saving.
If necessary, I can look for an old dump.
Regards

closed 2022-09-03 by DzikuVx