iNavFlight/inav #5126: Flip over after crash aka turtle mode

DangerD256, 2019-10-10:
Can't find such request here
Any plans to add it from betaflight?

marianomd, 2020-01-05:
This is particularly useful also when drone ends hanging from a tree.

Vasilifpp, 2020-02-19:
The last thing I miss on inav..

LeoNorw, 2020-04-17:
Just wanted to join in and support the request for Turtlemode.

evilphish, 2020-05-22:
Same here. Especially with a firmware like INAV that is sort of geared towards longer range flying this should be a must-have. I guess chances are that INAV users end up flipped over further from home far more often than others ;)

So: +1

TimJC513, 2020-06-02:
The lack of turtle mode is keeping me from using INAV on my 7" quad.

I'd also love to see dshot beacon supported, which would benefit multirotor and fixed wing.

gigapoc, 2020-06-04:
Same here, only thing preventing me to switch to iNav ! +1 !

charizardavi, 2020-07-03:
Same here. Would like turtle mode on inav!

almumen, 2020-07-10:
Is turtle mode still not available?

almumen, 2020-07-10:
> Is turtle mode still not available?

Sorry, just figured out from below progress bar that it is going to be a future development.

JohnRambo7039, 2020-10-10:
My plane crashed into the house three days ago, and it was belly up...
This is a very important function, hope can be achieved!!!

Vitaly591, 2020-12-15:
I need this feature.

UweHeim, 2021-02-06:
+1 to set higher priority on this. schouldn't be rocket science. is the slow approach on this to save memory for the still supported F3? or any other concerns? would be nice to see some kind of feedback on why this seems to have no significant priority since 2019...

avsaase, 2021-03-29:
This is resolved in #6685 so this issue can be closed.

closed 2021-04-03 by avsaase