iNavFlight/inav #5127: Board / Sensor alignment confusion in configurator / OSD elements

ShawnzAZ, 2019-10-11:
When the board is rotated 90 or 270 degrees in yaw, This skews what shows in the Configurator and OSD.  Roll and Pitch are reversed.

If I use adjustments tab to adjust board pitch angle in flight and have the board pitch and roll angles displayed, I will see it in the OSD present as roll adustment applied.   It shows the same in the 2.2.1 configurator.

In fact with the board rotated 270 degrees, if I input 45 degrees to pitch, it will display on the Setup tab as 45 degrees of roll.  To adjust board pitch, I need to enter it as roll in the configurator.   If I use in flight adjustments to Board angle pitch, it will apply the change to 'roll' in the Board and Sensor alignment section of the configurator as well and showing as roll in the OSD elements that I will display while adjusting board angle in flight.

Inav 2.2.1 on Matek F405-Wing (MF4S), INav configurator 2.2.1

# dump

# version
# INAV/MATEKF405SE 2.2.1 Jul  3 2019 / 22:48:46 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1
smix 4 5 0 50 0 1
smix 5 5 14 -125 25 -1
smix 6 6 0 -50 0 1
smix 7 6 14 -125 25 -1
smix 8 5 1 50 0 1
smix 9 6 1 50 0 1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 -100
servo 7 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 1 3 1 6 0 1200 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
logic 10 0 0 0 0 0 0 0
logic 11 0 0 0 0 0 0 0
logic 12 0 0 0 0 0 0 0
logic 13 0 0 0 0 0 0 0
logic 14 0 0 0 0 0 0 0
logic 15 0 0 0 0 0 0 0

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
[... 1106 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

DzikuVx, 2019-10-11:
In 'Configuration' tab, there is board orientation section when you have to set this up, not via adjustments. Just enter 90 or 270 and done

closed 2019-10-11 by DzikuVx

ShawnzAZ, 2019-10-11:
@DzikuVx you misunderstand.

The 90/270 works fine for the yaw adjustment.  What I'm saying is that to adjust pitch once rotated to 90 or 270, I have to make the adjustment to roll.   When using adjustments to do it in flight, adjusting board angle pitch will apply the changes to roll.

It functions fine, it is just confusing when you see your pitch adjustment input in the roll setting.

DzikuVx, 2019-10-11:
Adjustment are happening in sensor reference not uav reference. It's maybe
somehow misleading but consistent everywhere. Includes both configurator,
cli and adjustments

ShawnzAZ, 2019-10-11:
That makes sense.  Thank you!