iNavFlight/inav #5132: Dihedral on a plane

paralin, 2019-10-12:
## Current Behavior

iNav expects ailerons centered = level flight. On a plane with dihedral, this causes oscillation as inav attempts to maintain a roll angle. 

## Desired Behavior

The airplane needs to hold the control surfaces in a "roll left" configuration to counteract the self leveling tendency of the dihedral wings and maintain a constant roll angle. 

## Suggested Solution

This would be some sort of dynamic center adjustment according to current roll angle and desired roll rate (since the controllers are in terms of rates). 

Some sort of parameter for the "self leveling factor" of the aircraft that might be auto-tunable.

DzikuVx, 2019-10-17:
What you describe is Igain on PIFF controller on roll axis in this case.

paralin, 2019-10-17:
Thanks, I will test this.