iNavFlight/inav #5134: suggest add an new RTH mode for quadcopter

sincoder, 2019-10-13:
## Current Behavior
heading home ,and  RTH
## Desired Behavior
add a new RTH mode, force Quadcopter roll angle=0 , this help the quad RTH in big wind
## Suggested Solution
quadcopter force roll = 0 ,remember the point RTH start, then fly forward at NAV speed, at the same time, do a slow yaw adjust till the new GPS coordinate are at the line of home point to the RTH point,this will make the copter fly straight to home with a circle at begin   
## Who does this impact? Who is this for?
quadcopter
I lost my Quad due to too big wind ,the quad tilt to MAX roll(20) and Max pitch (18) , one motor overload ,and the Quad failed to hold it's attitude and lost a lot of altitude
## Additional context
with roll angle disable , the motor will hard be overload (always has two motor works for pitch angle ),and fly back home even the voltage was very low

Jetrell, 2019-10-15:
The function sounds workable. 

But its still no garentee that the quad will make it home if it isn't producing enough thrust to fight the headwind.

sin5678, 2020-02-04:
> The function sounds workable.
> 
> But its still no garentee that the quad will make it home if it isn't producing enough thrust to fight the headwind.

no, In this mod, quadcopters will always use two motors to provide sufficient tilt, Ability to withstand greater wind speeds

danarrib, 2020-02-04:
> I lost my Quad due to too big wind ,the quad tilt to MAX roll(20) and Max pitch (18) , one motor overload ,and the Quad failed to hold it's attitude and lost a lot of altitude

I guess that this request is not very clear on what exactly needs to be done.

You lost your quad because it has few bank angle authority (`nav_mc_bank_angle`, `max_angle_inclination_rll` and `max_angle_inclination_pit` variables)

You should give more room for INAV to bank more, this way INAV will have more authority to make the aircraft fight against wind.

It's important to note that INAV will always respect the speed limit, defined in `nav_auto_speed` variable... So in the case that there's no wind to fight, INAV will reduce the bank angle of the quadcopter in order to keep the aircraft on the desired speed... In other words: If you set the bank limit to 40 degreees, INAV will only keep the aircraft at 40 degrees to fight wind... If there's no wind, the actual bank angle will probably be a lot smaller than that.

DzikuVx, 2020-04-14:
Such approach will change nothing in terms of safety since nobody knows the wind direction. The problems were caused by poor thrust to weight ratio and low max angles. And probably some other hardware problems

closed 2020-04-14 by DzikuVx