iNavFlight/inav #5136: Launch mode enabled while flying causes problems

skorokithakis, 2019-10-13:
## Current Behavior

I just had a crash because the disarm switch got bumped in mid-air and I could not rearm due to launch mode (I forgot the mode enabled after I launched). #4948 is another ticket where launch mode causes problems.

## Suggested Solution

Unfortunately I have no proposal for the launch feature, but for the launch mode, it can be disabled after completion. So, the craft goes into autolaunch mode, it launches, launch is complete (either via timeout or stick motion), and autolaunch mode then becomes permanently disabled until the channel is moved to off and back on.

This would prevent forgotten switches from causing problems, as the craft would arm as normal. It might create a problem where someone forgot launch mode on while flying, landed, disarmed, then thought they could fly again without cycling launch mode, but this is rare (you normally check the switch and would remember to cycle it) and at most would cause the craft to flop down from a hand toss, rather than a crash from a large height.

FPVZaphod, 2020-01-27:
I have launch mode permanently enabled, and managed to disarm the plane in flight already. No issue at all, just mode elevator or aileron a little bit after arming, and you instantly have control back. Just need to remember the right order: 
Throttle to ZERO (or it wont arm), then ARM, then move Aileron/Elevator a bit, then Throttle up.

skorokithakis, 2020-01-27:
That's exactly the issue, though, that's a lot to remember when panicking with the aircraft racing to the ground.

FPVZaphod, 2020-01-27:
Will be not easy to change that without making the launch mode itself more complicated. 
There is not so much reason to panic, the plane is still controllable when disarmed. Motor is off, but everything else works. After a successful arm, you will anyways use the control surfaces, so there is no need to remember this step. Only thing to remember is to have Throttle Zero when arming. But this is not an autolaunch issue, it's a general thing, with all (?) Flight Controller Software, no matter if iNav, Betaflight or any other -flight. None of them will Arm when Throttle is not Zero.

skorokithakis, 2020-12-16:
I guess we can close this, if autolaunch gets disabled by stick commands without performing the actual launch. As you say, @FPVZaphod, I will use the control surfaces so hopefully that will disable autolaunch without an actual throw.

skorokithakis, 2020-12-16:
I just tested it and it looks like if you raise the throttle after arming, launch mode will stay enabled until you throw, which means that, in an emergency situation, you have to remember to move the sticks BEFORE raising throttle, which in my opinion is too much.

I think an improvement would be to disable launch mode with ANY stick input, even after throttle has been raised but before the throw. As you say, @FPVZaphod, I will certainly need to use the control surfaces after I arm, but since that doesn't disable autolaunch, I think this is still a problem that should be fixed.

avsaase, 2020-12-16:
I just had this happen to me and indeed it is really not a problem. Throttle to zero, arm, and you have control back.

skorokithakis, 2020-12-16:
@avsaase Not if you throttle up before moving the sticks.

avsaase, 2020-12-16:
So if you arm, raise throttle and then move the stick you do not get back control?

skorokithakis, 2020-12-16:
@avsaase Yes, if you have launch mode on and raise the throttle after you arm, the launch process starts, and moving the sticks does not give you control back. Does it for you? I can try again, maybe I made a mistake.

avsaase, 2020-12-16:
I would expect that the launch sequence would be aborted regardless of if you first raise the throttle or not. If it doesn't then I agree that should be changed.

skorokithakis, 2020-12-16:
I just retried and I can confirm: Sticks do nothing if you throttle up but do not throw. They *do not* cancel autolaunch.

FPVZaphod, 2020-12-17:
But it makes sense. If you have the plane on a bench, and you power it up with Autolaunch permanently enabled, then you arm, then you raise throttle, and then you move the control surfaces, it should NOT spin up the motor. This would be quite dangerous. 
For Motor Bench test, power up the plane, throttle zero, then arm. Then wiggle the right (Mode2) stick, and you can throttle up the motor. Same procedure in mid air. It won't arm with throttle up, and the motor will be only enabled if throttle is zero. 
And it's the only thing to remember when accidential disarm happens: Throttle down! For the safety of your fingers.
What else could be the solution? Maybe altitude based, or speed based? If speed > launch detect speed, then allow to spin up the motor after right stick wiggle even if throttle is not zero? But throttle has to be anyways zero to enable arming, so it wont help so much.
And allow arming with throttle > 0 if speed > launch detect speed  is way too dangerous.

skorokithakis, 2020-12-17:
If I have the plane on the bench, the props are off, and it doesn't matter what spins up or down. If it's on the bench and the props are on, I'm on my own. I don't think it's worth risking crashes to save people who bench test models with props on.

avsaase, 2020-12-17:
I hadn't thought about this but I agree @FPVZaphod. I also think that the problem should be solved at its root: disarming mid flight. I think setting `disarm_kill_switch` to OFF will be the best solution.

FPVZaphod, 2020-12-17:
But disarm happens also with the Arm/Disarm Switch, without using the Kill Switch.
What's about solving it like Betaflight - disarm only when throttle zero (as an option to choose)? Would at least cover some of the cases. Otherwise things become complicated. Disarm only when speed = 0 on a fixedwing? Or a similar way like the prearm-safety-override, where you can arm with no GPS-Fix when YAW is fully right while you arm?  Means as an example, disarm only possible when throttle zero and yaw fully right (or left) while you flip the switch?

avsaase, 2020-12-17:
From [Settings.md](https://github.com/iNavFlight/inav/blob/master/docs/Settings.md):

> disarm_kill_switch | ON | Disarms the motors independently of throttle value. Setting to OFF reverts to the old behaviour of disarming only when the throttle is low. Only applies when arming and disarming with an AUX channel.

skorokithakis, 2020-12-17:
> And it's the only thing to remember when accidential disarm happens: Throttle down! For the safety of your fingers.

This is wrong. You *also* need to remember "move sticks before you throttle up ONLY IF autolaunch is enabled", which is way too much to remember in an emergency situation. There's no reason for autolaunch to not get disabled with stick input at any point.

> I also think that the problem should be solved at its root: disarming mid flight.

That's even worse. Then you'll potentially get a model with a spinning prop running into the ground (or worse, people) and need to remember to throttle down in addition to disarming.

FPVZaphod, 2020-12-17:
So - problem mostly solved? At least if not flying a glider :-)
If you manage to disarm with throttle zero in mid air, then you can just arm again. Then wiggle the right stick to disable autolaunch, raise the throttle and go ahead...

And no, it doesnt matter if you have autolaunch enabled or disabled. Throttle zero, then arm, then wiggle will always bring the bird back alive. No other version required to remember.

skorokithakis, 2020-12-17:
On top of all this, there's no valid situation where I'm moving the sticks when in autolaunch and *want* them to be ignored. When I'm moving the sticks, I want control.

FPVZaphod, 2020-12-17:
Not always. Sometimes you accidentially move a stick while throwing the plane. That's why there is the parameter to ignore autolaunch abort if distance from RTH point is smaller than a certain value.

skorokithakis, 2020-12-17:
> Should be easy and safe, maximum requirement is throttle zero. And no matter in which order. Throttle zero then disarm has to be same like disarm and then throttle zero.

I don't understand what's so hard about this: I don't want to have to remember two versions. Usually it's "arm and do whatever you want", but if you have autolaunch it's "arm, move the stick before you throttle up, then do whatever you want", and there's no reason for autolaunch to ignore stick input.

I believe I have adequately explained my reasons, if you still aren't convinced and would rather have autolaunch ignore stick input because someone might have the model armed on the bench with the props on, that's not something I'll ever agree on.

FPVZaphod, 2020-12-17:
Currently you can choose it. That's the best option.
You can ignore stick input up to a certain distance, or you can enable it from the beginning.
But for no reason it should be possible to enable a motor when throttle is not zero. Enabling it when the first command to the ESC will be a high value should not be possible under any circumstance.

FPVZaphod, 2020-12-18:
> I hadn't thought about this but I agree @FPVZaphod. I also think that the problem should be solved at its root: disarming mid flight. 

Thinking a night over the problem - I don't think this is the only root of the problem. @skorokithakis said, he doesn't like to remember two different ways to regain control. Two reasons against this:  1. If you enable something permanently (autolaunch), then you need to live with the side effect, that you have to disable it again in the case you don't need it.  2. You don't need to remember two versions, because the version "Throttle Zero, Arm, Stick Wiggle, Raise Throttle" covers all. Even if autolaunch is not permanently enabled, this will bring back your plane to life. So only one sequence to remember.
But the root problem is not the possibility to disarm in-flight, there are some cases where this might be possible instantly. The root problem is, that "permanent enable autolaunch" should be intelligent enought to detect, if there is no launch possible, because the plane got launched already. So after powering up the plane and arming, a check should verify, if a launch is required. Could be as simple as a speed check, or a check, if the motor was already spinning with minimum 50% throttle after the power-up. 
Not nice would be the possibility to only have autolaunch after the first arm. I used to power up my plane, arm it, disable / stick-wiggle autolaunch, perform a quick ailerion / motor check, then disarm again, re-arm to enable autolaunch again, and then go. 
So what's about a check, if the motor was running (ESC Command > 50%) already since power up, and if yes, then don't activate autolaunch again?

closed 2024-02-02 by DzikuVx