iNavFlight/inav #5138: Spektrum Receiver SPM 4649t & Failsafe not working

kdigit, 2019-10-13:
Hi,

I´m new to quads and not able to getting the failsafe feature to work. My setup is a Spektrum DX6 Black + SPM4649t, with a skystars 7" copter. GPS is working properly. And the quad is flying great. 

But today I had a signal loss because my DX6 TX suddenly just shuts off...:( I thought, glad I´m programmed the RTH failsafe....But nothing happens the quad just crashed....Luckily only the props are damaged...Anyway..

The FC obviously is not recognizing that there is a failsafe moment because my guess is that the Spektrum RX is sending any signal though the TX is out of signal. The manual of the RX says that it has no failsafe function the FC has to do this. But if the RX has no feature like "no pulses" how can the FC recognize there is a failsafe event?

Are people here with the same Spektrum Receiver and a working failsafe setup? Can anyone give me an advice for these issue? How can I tell the FC there is no signal?

Any help will really appreciated.
Thanks

Andy

Jetrell, 2019-10-14:
You will have to set the receives failsafe HOLD (in your receiver manual). To output the correct channel positions when the TX/RX link is lost. 
You need to set the SPM 4649t so that it outputs the correct channel to trigger iNav's `Failsafe `mode for a return to home condition.

[image]

All that effectively does is trigger your RTH switch automatically when a signal loss occurs.  Spektrum does not output failsafe data for iNav to use to trigger RTH, in the way you described.

kdigit, 2019-10-14:
Hi Jetrell,

thanks a lot for your reply. Yeah, you´re absolutely right the function HOLD would be the solution. But I´ve red the RX manual about five times (it´s a very short manual, by the way) and there is absolutely nothing about HOLD or NO PULSE. 

I think I have to contact the Spektrum Support for this question. Or maybe someone has the same RX/TX setup?
Thanks again for your explanation. I will report...

Regards 
Andy

Jetrell, 2019-10-14:
@kdigit The Spektrum system has two FS. `Hold last command FS` and `Preset FS`.
You want Preset failsafe.  
https://www.horizonhobby.com/pdf/SPM6700-Manual_EN.pdf page 12 explains it.

kdigit, 2019-10-14:
Oh, wow. I thought I did a quick search thru the TX manual but it seems I´ve overlooked that important page.
I now found a youtube video that explains that exact Fs modes because the manual delivers not in depth information...;)

Thank you again for your help I really appreciate that.
Regards
Andy

closed 2019-10-15 by kdigit