iNavFlight/inav #5139: Landing runway definition for automatic landing

xkriton, 2019-10-14:
## Current Behavior
With automatic landing of fixed wing there is random direction of final landing after descending near home point

## Desired Behavior
Landing on defined runway in straight line

## Suggested Solution
- Add choice this for automatic landing mode (also for failsafe RTH)
- Define two endpoints of runway (maybe as special waypoints in map?)
- Decide landing direction against calculated wind direction (if enabled) after descending to pre-landing altitude (or from nearest endpoint when: wind at approx 90 degrees or wind direction not calculated)
- Approach landing line from chosen endpoint and land pointing in direction to the other endpoint (to keep on defined runway line)

Optional extension:
- Define multiple landing runways to memory and chose nearest runway after first arming based on shortest distance from home point.

## Who does this impact? Who is this for?
Fixed wing users

## Additional context
N/A

giacomo892, 2019-10-14:
We have this in the works:
https://github.com/iNavFlight/inav/pull/4961

closed 2019-10-14 by giacomo892

xkriton, 2019-10-14:
Sure, I have seen it. But will it hit desired runway or just land somewhere around in preset direction? Will it consider wind direction?
Intention of this proposal (or as extension of #4961 ) is that when I am flying with others and we have people and cars 15-20m and tree line away along our runway, it would be good not to hit them in case something goes wrong.
With loiter circle greater than this distance and not "precise +/-5m"  landing along runway, there is good chance it can hit something with just simple landing vector.

Something like this is used in MFD Crosshair as basically shown in this video https://www.youtube.com/watch?v=EPk02NJwSec