iNavFlight/inav #5143: Feature Request: GPS Output on spare UART

wx4cb, 2019-10-15:
## Current Behavior
Not Available

## Desired Behavior
The ability to mirror the input from the GPS onto a spare usart (or softserial) so that we can have an NMEA or UBX output to something such as an MFD telefly module. That way we don't need to have 2 GPS receivers or tee off (with possible buffer and/or level issues on the receive line) if we wish to run a secondary module for an antenna tracker or even something like a raspberry pi zero for camera control of a second mapping camer that way we can embed the gps data into the EXIF for the photos

## Suggested Solution
The ability to use a spare uart or a softserial for gps output.

## Who does this impact? Who is this for?
This can benefit all users especially us users that like mapping with a secondary computer and also those who use something like an MFD antenna tracker which requires a separate module with a separate gps input to embed gps data into the VBI section of the video frame. 

This should be relatively simple as the gps input is already there it should also send all gps data to the mirror port irrespective if there's enough satellites etc. The telefly module can understand both NMEA and UBX data and it only needs to be "receive only" (ie from FC to module) there is no need to be able to "talk" to the GPS.

wx4cb, 2019-10-15:
here's a link to the module manual: http://www.myflydream.com/inc/lib/download/download.php?DId=60

rrk1, 2019-10-16:
IMHO, making a simple 'ghetto' buffer is so trivial that it does not worth wasting dev's precious time. Most of a time a single npn transistor as emitter follower + about 5.1K load resistor will do okay.
Also, don't think you are allowed to do anything with the processor output :) Shorting it or injecting significant noise into it can do bad things, too, potentially crashing the aircraft.

wx4cb, 2019-10-16:
@rrk1 not sure what you mean by the "processor output"  

As far as the buffer goes, i know its trivial, but imho, using for example, the f722 wing like i do, has more than enough uarts to do it and that way you dont have to do anything "ghetto" and everything is at the right levels etc.

wx4cb, 2019-10-18:
So just tested "teeing" the gps off and all i get is gps failure every couple of seconds. I really need this, but not sure how to code it

tigert, 2019-10-28:
The first sentences of the TeleFly manual you linked explain a wiring scheme where you share the GPS TX and power wires with the TeleFly. That of course might have possible voltage mismatch issues with the different modules you were concerned about. So a  fair point.

There is an issue about implementing NMEA or UBLOX as telemetry: #4693 

Additionally, many projects like antenna trackers use LTM, MavLink or MSP protocol. There are likely libraries for Arduino to read and write these, so if you want to do such a thing with an on-board Arduino, I'd go with a telemetry link instead of a NMEA hack, as you could also make the arduino aware of arm status for example, which might be useful.

tigert, 2019-10-28:
@wx4cb just out of curiosity, what GPS is this? 

While looking at the nmea telemetry issue, I found #4688 where someone had repeated gps failures that prevented a tracking device from working. The root cause was INAV expected faster position updates from the GPS which resulted in INAV repeatedly reinitializing the GPS connection, and while this "worked" as far as INAV being more or less aware of its position, a tracker device connected to the same wires got confused by the repeatedly reinitialized GPS connection.

Could this be why you have issues too?

digitalentity, 2019-10-28:
Closing as a duplicate of #4693

closed 2019-10-28 by digitalentity