iNavFlight/inav #5144: Throttle Hold Mode - Fixed Wing

breadoven, 2019-10-16:
## Current Behaviour
There is no obvious means of stopping the motor on a fixed wing in flight, regardless of throttle stick position, without disarming or using the Throttle Hold function on the transmitter (which disables the throttle stick output meaning it can't be used for other functions). Disarming interrupts other iNav functions, e.g. Blackbox, VTX power control, OSD etc, so is undesirable in flight.

## Desired Behaviour
Function added that stops the motor whilst armed regardless of throttle stick position (Throttle Hold function).

## Suggested Solution
Add a Mode that stops the motor whilst still Armed regardless of throttle stick position. Potential safety issues could be mitigated by only allowing Mode to be turned off when the throttle is set to low, as with arming, i.e. the Mode is latched on and can only be turned off when the Mode command is off and the Throttle is low.

A more flexible solution would be to change Throttle parameters (probably **min_check** and **max_check**) to change the throttle control range to preset values specifically for that Mode reverting back to default values when Mode is off). This would move the effective control range of the throttle stick to a different range of stick movement, e.g. increase min_check to squeeze throttle control range to the top part of the stick movement range whilst cutting the throttle below allowing the lower range to be used for flaps etc without the motor running. This would effectively provide Throttle Hold in the lower range of stick movement but allow the motor to run if needed by pushing the throttle stick to high (sort of emergency power).

## Who does this impact? Who is this for?
This would be for fixed wing only and mainly for motor gliders where the motor may not be required (when soaring, thermalling) but the throttle stick is still required to be used to operate Flaps or Spoilers etc (Via transmitter mixing or iNav).

## Additional context
None

DzikuVx, 2019-10-16:
So, basically, only Throttle input into the mixer would be set to `0` and influence only the motor. Right?

breadoven, 2019-10-16:
Correct, this would only influence the motor. It's just a switchable way of stopping the motor running (without disarming) so the Throttle stick can be used to do something else such as controlling Flaps. Pretty well restricted to use on motor gliders probably.

DzikuVx, 2019-10-18:
Will be part of INAV 2.3, but no GUI, CLI will have to be used to setup this function

closed 2019-10-18 by DzikuVx

breadoven, 2019-10-18:
I was looking at the logic conditions but quickly realised they only worked for servos not motors. This tweak should do the job I was thinking of. So thanks for that, quick work !

PR #5147 (2019-10-18): Allow to override mixer throttle by global function
Fixes #5144