iNavFlight/inav #5146: restricted telemetry

Signellox, 2019-10-17:
Taranis X9D with FrSky R9Telemetry Receiver and a Matek Systems F405-WING (New) STM32F405 Flight Controller Built-in OSD .

Taranis detects only RxBt and RSSI Sensor. (?)
All other functions seems to be ok.

S Port is conected and activated.

I reinstalled the INAV software but its not completly clear because there are four posibilities.
MATEK 405 did not work, MAtek 405SE, MATEK 405 SERVOS, MATEK 405OSD.
I installed MATEK 405 OSD.

It would be nice to get some help on this topic.

Pairan, 2019-10-17:
You should take Matek 405SE as a Target ... in case you dont know which target to take, then go to the website of Mateksys and have a look- they'll tell which Target is to be used!

Signellox, 2019-10-17:
Thanks I reinstalled MATEKF405SE and activated telemetry output (inav configuration).
I have done a new binding process for the reciever and started "discover new sensors".
But Taranis detects only RxBt and RSSI Sensor.

Pairan, 2019-10-17:
did you activate telemetry in the configurator?

Signellox, 2019-10-17:
Yes, telemetry ist activated in inav configurator. Thanks.

DzikuVx, 2019-10-17:
This video covers the setup: https://www.youtube.com/watch?v=-cu8gO9px8M&list=PL9ISXGL4cHgebl3b4JXTW9Z_LrhY7Lu2G

Signellox, 2019-10-17:
Thanks together for response. The video is good. And setting is exactly in the same way. Beside only using SPort. I have had the full functionality before a crash. After the crash of the airplane  reconfigured the whole system and the problem occured.
- I tested also my second receiver.
-  I have a secon board F411 on a different plane. This works whithout problem.
- In the OSD display of the googles I the artifical horizon works and the homedistance is correct.
- On the PC display i can not see any failures besid by using Canal 16 for RSSI sometimes it works sometimes not.
- but taranis can not discover the sensors.

Signellox, 2019-10-17:
i checked the wiring again its ok.
Short:
everthing works normal  besides taranis did not get the sensors

Signellox, 2019-10-18:
ty. i will change to my other controller F411 and try again.

closed 2019-10-18 by Signellox