iNavFlight/inav #5152: iNav refuses to spin motor even though "armed", servo gets "sticky"

SquashSquishSquaw, 2019-10-20:
Hi,

not sure if this is a bug, or a feature, or just a pretty stupid user mistake?

My plane runs iNav on a Matek F405-Wing.

I had just landed, with cell voltage at about 3.85 Volts. I'm not sure if I had disarmed in between or not, but when trying to fly again, I was not able to spin up the motor any more.

I then disarmed, armed again ("motors armed" as reported by the LUA script), and still had the same problem. No matter what I tried, spinning up the motor was not possible. Switching to different flight modes didn't change a thing.

At the same time, I noticed that when I pulled the rudder stick and released it soon after, the rudder would "stick" a tiny bit away from neutral position, in direction of where I had pulled to. In other words, the rudder did not completely return to neutral. It was away from neutral by an estimated 5 or 10 %. This was the case no matter which direction I pulled the rudder into. Pull it to the left, and it sticks left side, pull it to the right again, and it sticks to the right side. This wasn't just servo or rudder play, the servo was really trying to hold that "off neutral" position.

After rebooting the FC, everything was back to normal again and I was able to fly as usual.

On the next few landings, I didn't have this problem any more.

Now I'm having a kind of a nervous feeling. Any ideas or suggestions please? Many thanks in advance for your time!

Jetrell, 2019-10-22:
I have experienced a similar thing with re-arming from time to time. Its not a bug, just a specific sequence of things that need to occur to re-arm. 
Here is a link - https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons

As for the Rudder or any other stabilized control surface. Its not uncommon for the control surfaces to wander a little from center in `Acro/Air `mode, while being held or even sitting on the ground. 
If you switch to `Manual` mode it will center.

SquashSquishSquaw, 2019-10-22:
Hi,

and many thanks for your answer.

Rearming hasn't been the problem, the Frsky LUA telemetry script clearly told me that the motors are armed. Also, I was able to get into RTH mode. On the other hand, even though the plane was armed, I was not able to spin up the motor.

https://github.com/iNavFlight/LuaTelemetry

The rudder was being "sticky" no matter which mode I switched to. I tried switching to manual mode and back, and I kept having this problem until I rebooted the FC.

Jetrell, 2019-10-22:
Its not impossible for a glitch to occur. Of which a reboot fixed.
Do you have any `log` recordings of that event? It could possibly help track down the issue.

SquashSquishSquaw, 2019-10-23:
Yes, it will take a few days though since I've just ordered a SD card reader for this purpose. It seems that there is no other way to read SD cards from the flight controller, without destroying the contents?

DzikuVx, 2020-04-14:
seems like stalled, closing

closed 2020-04-14 by DzikuVx