iNavFlight/inav #5153: kakute f7 V1.5 dont works with inav

multinnov, 2019-10-21:
hello,
My kakute F7 V1.5 dont work with inav.
This is the message when a try to connect the kakute F7 V1.5 to inav. 
[image]

With an other older kakute V 1.0 it works.
Is the Kakute F7 V1.5 normaly work with inav? or the firmware is not yet done?

Can you help me, Thanks

Pairan, 2019-10-21:
... perhaps you need to flash iNav on the Kakute to make it work with iNav ... !

multinnov, 2019-10-21:
Flash no work too. I think so, INAV have not add Kakute F7 V1.5  (V2 name on betaflight and work).

DzikuVx, 2019-10-30:
@multinnov I do not have Kakute F7 1.5 but I think I localized the problem. Could you check if firmware for attachment works?
If yes then great, if not we will have to wait for me getting my hands on the physical board
[inav_2.2.1_KAKUTEF7.zip](https://github.com/iNavFlight/inav/files/3791120/inav_2.2.1_KAKUTEF7.zip)

multinnov, 2019-10-31:
Howww, thank you very much !!!

 

It is good 😊

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

DzikuVx, 2019-10-31:
@multinnov just to confirm: KakuteF7 rev 1.5 from the hex file is working fine, right?

multinnov, 2019-10-31:
Yes, perfect, it work .

 

 

thanks

 

https://www.avast.com/antivirus

DzikuVx, 2019-10-31:
#5179

closed 2019-10-31 by DzikuVx

MaxLansing, 2019-11-11:
@DzikuVx Hi Mate, i see you closed this thread, but if i may, i would like to ask you a favor.
As you provided the only working hex file for the Kakute F7 V1.5 Board, for which i am really thankfull, i would like to provide some testing. i am new to INAV, but i built a prototype based on the Holybro F7 V1.5 + Tekko F3 + Atlatl Stack. Thank to your HEX, i was able to flash INAV onto it and get it running. But i can't select any other then Standard ESC Protocols, since they are most likely disabled in the build. 
Therefore my Question:
Could you compile another build with all ESC protocols enabled? i would be glad to try them out and provide feedback if you want. 
Thank you in advance, best regards, max

DzikuVx, 2019-11-12:
In the hex use Multishot or flash INAv 2.3RC1 that should have DSHOT as well

MaxLansing, 2019-11-12:
Well, i admit it, i am stuck at this point, on this advise. The board is not an official target, so i don't know what version to flash onto it since i have to choose a board first.

DzikuVx, 2019-11-12:
Flash kakute f7 target to your kakute F7 board after selecting 2.3 RC1 from the list when "show unstable releases" is on