iNavFlight/inav #5159: Support for Matek F411 Rx Board

emergent-idg, 2019-10-25:
I am working with autonomous flight and micro/mini UAVs.  I am trying to flash iNav to the EMAX TinyHawk Freestyle which utilizes the Matek F411 board.  However, all of the supported firmware options, once flashed, will only recognize two of the ESCs/Motors.

When I reflash Betaflight with the Matek F411RX Firmware, recognition of all four ESCs and motors is restored.

Since iNav already supports the Matek F411 board, would it be possible to add support for the F411RX?

lavachemist, 2020-09-11:
I'm also interested in this. Would be awesome to have iNav on the Tinyhawk quads.

gcl2000, 2021-03-07:
Me too. Can anyone
Confirm what target in INAV will this FC work? Thanks

DzikuVx, 2022-05-07:
Board is no longer manufactured

closed 2022-05-07 by DzikuVx

jonathanvolker, 2023-05-30:
good morning, is there no inav support for this board? I would like to build a wing with this board, I already adapted the servos but with betafligth I can't make it work correctly, it tries to stabilize even if it's in acro, that's why I want to try Inav.

LinkinL0gs, 2023-11-08:
@DzikuVx While the board may no longer be manufactured for individual hobbyist parts it's definitely still in use by EMAX in their Nanohawk and other hawk quads and perhaps other manufacturers.  Was curious to see for myself today if inav supported this.  see https://emax-usa.com/products/nanohawk-spare-parts-aio-flight-controller that's listed as being target: F4（MATEKF411RX)

DzikuVx, 2023-11-08:
@LinkinL0gs we no longer accept any F411 based targets