iNavFlight/inav #5165: BLheli_32 ESC "Auto Telemetry"

Jetrell, 2019-10-26:
This issue references #2710 
The flight controller normally requests telemetry data from the ESC. 
But if `Auto Telemetry` is enabled in the BLheli_32 suite. The ESC will send out telemetry data to the FC at set intervals. 
This is useful if we want to use an analog ESC control protocol(multishot), instead of digital(Dshot).
This feature is more applicable to Fixed-wings, than Multi-rotors. Because the data from multiply ESC, being sent at the same time may get corrupted.

[image]

digitalentity, 2020-05-04:
Damn, I forgot about this one :scream:

Jetrell, 2020-05-04:
@digitalentity 

> 
> 
> Damn, I forgot about this one 

These things happen :-)

digitalentity, 2020-05-05:
#5674 created

PR #5674 (2020-05-07): [ESC_SENSOR] Add listen_only mode
Fixes  #5165
closed 2020-05-05 by digitalentity